GPUOpen-Drivers/AMDVLK #169: GTA IV refuses to launch with AMDVLK

kattjevfel, 2020-07-11:
Game launcher runs but main application crashes instantly, runs perfect with RADV.
[steam-12210.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/4906872/steam-12210.log)

Arch Linux
AMD Ryzen 5 1600X
AMD Radeon RX Vega 56
16 GB RAM

amdvlk 2020.Q3.1
llvm 10.0.0
Proton 5.0-9

Launched with: `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd32.json:/usr/share/vulkan/icd.d/amd_icd64.json PROTON_LOG=1 %command%`

Flakebi, 2020-07-13:
As you are using arch, maybe you are hitting #165? amdvlk does not seem to work with mesa > 20.0.7-3 on some systems.

kattjevfel, 2020-07-13:
@Flakebi It works just fine on some other games, both DXVK and native ones.
To name a few: Spyro: Reignited Trilogy, My Time At Portia, Divinity: Original Sin 2 and vkQuake.

Flakebi, 2020-07-13:
Ok thanks, seems like something different then.

JacobHeAMD, 2020-07-16:
@kattjevfel , did you uninstall radv before trying with AMDVLK?

kattjevfel, 2020-07-16:
@JacobHeAMD No I kept RADV installed and launched the games with `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd32.json:/usr/share/vulkan/icd.d/amd_icd64.json PROTON_LOG=1 %command%` which works just fine on other games, is this not the right way to do it?

massatt212, 2020-07-29:
i wonder if AMDVLK will ever use ACO or can it ?

JaxLinAMD, 2020-07-29:
AMDVLK can't use ACO for now and I don't think there are any plan of using ACO so far.

jinjianrong, 2020-08-19:
Reproduced the issue with Grand Theft Auto IV. It is a regression

JaxLinAMD, 2020-09-15:
@kattjevfel issue fixed in v-2020.Q3.5, please verify with it.

closed 2020-09-16 by jinjianrong

kattjevfel, 2020-09-16:
@JaxLinAMD I tested Fallout 76, My Summer Car, FlatOut: Ultimate Carnage and GTA IV. Both F76 and MSC worked, but not the rest, the ones working being 64-bit, and the rest are 32-bit and refused to launch.

GTA IV: [steam-12210.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/5233715/steam-12210.log)
FlatOut UC: [steam-12360.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/5233717/steam-12360.log)