GPUOpen-Drivers/AMDVLK #174: GPU hangs on RX580, v-2020.Q2.6

LekKit, 2020-07-17:
I have GPU hangs while running several Vulkan titles with AMDVLK v-2020.Q2.6:

## Doom 2016
Can't even get into the game, after loading the screen freezes, then flashes and corrupts
This issue is not present when using AMDGPU-PRO or Mesa RADV, neither on Windows

## Wolfenstein 2
I can play for some time without problems, however occasionally it hangs (screen freezes then corrupts), same for AMDGPU-PRO

##### dmesg on GPU hang:
[ 1220.113265] [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
[ 1220.123159] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=994401, emitted seq=994403
[ 1220.123232] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process DOOMx64vk.exe pid 5399 thread DOOMx64vk.exe pid 5434
[ 1220.123239] amdgpu 0000:09:00.0: GPU reset begin!
[ 1220.768678] cp is busy, skip halt cp
[ 1221.072352] rlc is busy, skip halt rlc
[ 1221.073377] amdgpu 0000:09:00.0: GPU BACO reset
[ 1221.380936] amdgpu 0000:09:00.0: GPU reset succeeded, trying to resume
[ 1221.383019] [drm] PCIE GART of 256M enabled (table at 0x000000F400900000).
[ 1221.383035] [drm] VRAM is lost due to GPU reset!
[ 1221.762558] amdgpu 0000:09:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring comp_1.2.0 test failed (-110)
[ 1221.874174] [drm] UVD and UVD ENC initialized successfully.
[ 1221.974221] [drm] VCE initialized successfully.
[ 1221.978825] [drm] recover vram bo from shadow start
[ 1221.985590] [drm] recover vram bo from shadow done
[ 1221.985592] [drm] Skip scheduling IBs!

##### CPU: Ryzen 7 1700
##### GPU: RX 580
##### OS: tested on Debian Sid, Debian Buster - same results
##### Kernel: tested vanilla 5.6, 5.7 kernels as well as xanmod 5.6

LekKit, 2020-08-20:
Freshly compiled GPUOpen-Drivers/llpc@44a8e50 seem to fix the issue with Doom 2016 (looks like it also fixes #175), testing Wolfenstein 2 now (this might be unrelated to LLPC, happens on closed-source AMDGPU-PRO driver too)

jinjianrong, 2020-10-21:
There is a fix for Wolfenstein 2  in Q4.1 release. Please check it

closed 2020-10-21 by jinjianrong