GPUOpen-Drivers/AMDVLK #175: Doom 2016 crashes after a while

ZerataX, 2020-07-23:
Playing Doom 2016 with RADV works perfectly fine and even amdvlk seems to work great with improved performance until you reach a certain point in the game when it suddenly crashes. most of the time when starting a new campaign it happens when I pick up the helmet.
When the game crashes the music still continues and I can see my mouse, but I can't alt-tab and after a while the frozen image gets replaced by something like this:

[image]
after a while I press the power button which then results in this:

[image]

I sometimes recover the session like this and the x-server restarts.
I tried starting from different saves I had and with different resolutions and settings, but it always crashes a few minutes in.

Also maybe interesting, but Doom eternal doesn't seem to crash, but I've only yet tried 30minutes.

I can't use renderdoc when the game crashes, but I can upload a capture from before it crashes if that helps.
here is my steam proton log and vulkaninfo output:
[vulkaninfo and protonlog.tar.gz](https://github.com/GPUOpen-Drivers/AMDVLK/files/4963378/vulkaninfo.and.protonlog.tar.gz)
I have [mangohud](https://github.com/flightlessmango/MangoHud/) installed, which might be reported in these logs, but deinstalling it doesn't seem to change anything.

```
------------- 
OS: NixOS 20.09 (Nightingale) x86_64 
Kernel: 5.4.51 
DE: Plasma (x11)
CPU: Intel i7-9700K (8) @ 4.900GHz 
GPU: NVIDIA P106-100 
GPU: AMD ATI Radeon RX Vega 56/64 (amdvlk 2020.Q3.1)
```

ZerataX, 2020-07-29:
update: 2020.Q3.2 still behaves exactly the same

jinjianrong, 2020-08-04:
It is a regression caused by [PR650] Rationalized user data usage

ZerataX, 2020-08-15:
seems like with 2020.Q3.3 this is fixed? I've played for roughly 30min and no crashes, so I'm going to close this issue now

ZerataX, 2020-08-16:
Sorry that was wrong, I accidentally deleted the amdvlk icd file so I just started doom with radv instead of amdvlk. This issue is obviously still present with 2020.Q3.3

jinjianrong, 2020-08-17:
Fixed in llpc dev branch with https://github.com/GPUOpen-Drivers/llpc/commit/44a8e50f5211fd718a2724e88e827e2e952d6fbf

JaxLinAMD, 2020-08-20:
GPUOpen-Drivers/llpc@44a8e50 will be promoted to master branch at v-2020.Q3.5( not v-2020.Q3.4)

JaxLinAMD, 2020-09-15:
@ZerataX  issue fixed in v-2020.Q3.5 please verify that, thanks!

closed 2020-09-16 by jinjianrong

ZerataX, 2020-09-17:
yay it does work! and also 10% faster than radv so that is also super cool
https://flightlessmango.com/games/6406/logs/735