GPUOpen-Drivers/AMDVLK #176: GUI settings app

jqadev, 2020-08-04:
Dear Team,

I remember when a few years ago I used pre-GCN Radeon GPU with fglrx driver containing Catalyst Control Center.
It was a very nice GUI settings app, similar to current NVIDIA X Server Settings but offering way more options.

I found two screenshots with this app:
https://www.blackmoreops.com/wp-content/uploads/2013/10/Install-AMD-ATI-Driver-fglrx-in-Kali-Linux-1.x-blackMORE-Ops.jpeg
https://www.phoronix.net/image.php?id=amd_radeon_hd7950&image=amd_hd7950_catalyst_show

Is there a chance for a release of Radeon Software Adrenalin or Catalyst Control Center again for Linux?
Maybe there was already a discussion regarding that and there is explained a reason why AMD dropped GUI settings app for Linux?

JacobHeAMD, 2020-08-19:
No plan

closed 2020-08-19 by JacobHeAMD

jqadev, 2020-08-20:
@JacobHeAMD please look at this https://github.com/NVIDIA/nvidia-settings
It's under GNU General Public License v2.0.

It would be great to have this panel for AMDVLK-FreeSync-VRR management for example https://www.phoronix.com/scan.php?page=news_item&px=AMDVLK-FreeSync-VRR-Option

It's 2020. Previously we had GUI settings app and now we need to use CLI when Linux finally became a great platform for gaming.

What do you think about AMD official fork of this project?