GPUOpen-Drivers/AMDVLK #177: slow CPU performance in some titles with DXVK

aufkrawall, 2020-08-16:
Originally reported here: https://github.com/doitsujin/dxvk/issues/1718

CPU performance is rather low with amdvlk-pro 20.30-1109583 driver (crashes with open amdvlk 2020.Q3.3-1 driver from Arch repo, so I couldn't test performance with it), no difference between Windows and Linux:
[image]
It's 70-80fps with amdvlk-pro, 120-130 with native D3D9 and 170-180 with RADV. 

You have to disable Unreal Engine frame rate smoothing in the game's config file to get unlocked fps.

### Software information
Mirror's Edge on Steam Proton 5.0, 1440p max. details (no vsync, MSAA nor PhysX)

### System information
- GPU: RX 480
- Driver: amdvlk-pro 20.30-1109583 / 20.8.1
- Wine version: wine-tkg-staging 5.14 / Windows 10 2004
- DXVK version: dxvk-master.d92660923acef5066711af43b8afa3196026eb3c

### Apitrace file(s)
https://drive.google.com/file/d/1DqS-TDmqoOnRuxMXzF4BSjgxJ4OOUuMn/view?usp=sharing

### DXVK Log files
[MirrorsEdge_d3d9.log](https://github.com/doitsujin/dxvk/files/5056813/MirrorsEdge_d3d9.log)

aufkrawall, 2020-09-13:
To provide you with a hint how to find the corresponding config file of the above mentioned Mirror's Edge game: It's

 `~/.local/share/Steam/steamapps/compatdata/17410/pfx/drive_c/users/steamuser/My Documents/EA Games/Mirror's Edge/TdGame/Config/TdEngine.ini`

There, change `bSmoothFrameRate=FALSE` to unlock fps (keep vsync disabled). Just start a new game to render the scene shown in the screenshot above. It's easy to replicate. Drop me a note if you'd like to have a renderdoc/apitrace capture.

user1-github, 2020-09-16:
From my observation, I think this has something to do with the combination of AMDVLK (both open and pro) + DXVK + Unreal Engine 3/4, because if you look at the latest [Phoronix benchmarks](https://www.phoronix.com/scan.php?page=article&item=amd-drivers-sep20&num=2), you can see that Batman: Arkham Knight, which is also an Unreal Engine 3 title, is much slower on AMDVLK drivers than radv. 
I tested myself Splitgate: Arena Shooter, which is an Unreal Engine 4 title and I get 20-25% lower fps on AMDVLK drivers than on radv/aco.

aufkrawall, 2021-01-06:
Seems this got fixed in recent drivers, I can't reproduce anymore on either Windows nor Linux with both amdvlk-open/pro.

closed 2021-01-07 by jinjianrong

aufkrawall, 2021-01-07:
Just wanted to be sure:
amdvlk-2020.Q4.5: low CPU performance
amdvlk-2020.Q4.6: ~3x faster CPU performance :)