GPUOpen-Drivers/AMDVLK #183: [Feature Request] Being able to force msaa

Toadfield, 2020-09-01:
Some games,new and old,don't have build in msaa,and the only options for aa I know is post processed,like fxaa ans smaa,nut they don't look really good in my opinion,so it would be fantastic to be able to force msaa,because it's the best looking aa option in my opinion,the edges of objects look really realistic and it doesn't need as much power as ssaa.

Degerz, 2020-09-04:
It won't work with Vulkan because we can't know if a game is actually rendering to the swap chain image or not so the best solution is you either make hacks specific for each game or have the developers support the feature natively ... 

The option to force MSAA in driver settings stopped working a long time ago so this idea would only be viable on really old games ...

Toadfield, 2020-09-04:
That doesn't sound good,because game developers often don't do what the players want,and per game hacks are nearly impossible to find for many games,and I'm to dumb to make them.
Btw how does the msaa forcing in the windows amd drivers work?
Because before I switched to Linux I could force msaa in the amd drivers and it worked in any game I tried and I didn't had any problem,so why does it work there?

Degerz, 2020-09-04:
It's mostly dependent on the gfx API ... 

Most of the time forcing MSAA on D3D10/11/12 or on Vulkan doesn't work for AMD or any vendor for that matter because only the developer controls whichever render target the geometry is being rendered to ... 

On many older D3D9 or OpenGL applications, most games actually only render to a single swap chain image so it's pretty easy for the driver to just force them to run at a different sampling rate ...

Toadfield, 2020-09-04:
Oh,that's why it worked for me in the past,I only played old games,and they used d3d9.
But is there a way to force msaa to games that are d3d9 in windows,but are translated to dxvk in Linux?

Oschowa, 2020-09-04:
You can force it in DXVK for d3d9 games
https://github.com/doitsujin/dxvk/blob/master/dxvk.conf#L322
but don't expect it to work for every game there as well.

Toadfield, 2020-09-04:
Thanks,but I tried it in Garrys Mod and it didn't work.
I created a config file here: ~/.config/dxvk/dxvk.conf
And I did that in the config file: https://i.imgur.com/CEWUc7b.png
So what did I do wrong?
Because I'm pretty sure that gmod uses d3d9.
Btw I didn't run the native Linux version,I ran it through Proton,so it must be d3d9.

Oschowa, 2020-09-04:
The dxvk.conf file needs to be placed either next to the games .exe or where the environment variable DXVK_CONFIG_FILE points to.

Toadfield, 2020-09-04:
I did the config into next to the exe,but if I start gmod,then I get this Error message:https://i.imgur.com/wnVRnrA.png
And I can't press "OK",because my mouse cursor doesn't work when this window pops up.

Toadfield, 2020-09-06:
I solved the problem,it didn't work because I tried to use msaa x16,but it only works with msaa x8 if I tested it right.
Now the game starts,but I don't know if it works,is there an easy way to check that?
And if I do "DXVK_CONFIG_FILE=1" into the launch option of gmod,and delete the file dxvk.config in the gmod folder,I'm pretty sure that it doesn't work,because it starts like normal with msaa set to 16.

Oschowa, 2020-09-07:
You are supposed to use DXVK_CONFIG_FILE like this: DXVK_CONFIG_FILE=/path/to/dxvk.conf

Toadfield, 2020-09-13:
Thanks,I tested it on some d3d9 games,and it worked for the most.
But is there a way to force ssaa for d3d9 games?
Because I want to see how good my gpu is.

Oschowa, 2020-09-13:
On AMD gpus you can use https://github.com/Plagman/gamescope for super sampling regardless of API used, but it's still WIP and you'll have to compile it yourself. Anyways, i think this issue should be closed.

Toadfield, 2020-09-13:
thanks,I will take a look on that.

closed 2020-09-13 by Toadfield