GPUOpen-Drivers/AMDVLK #184: AMD Ryzen 5 Pro 4650U with kernel 4.19

paykocaydin, 2020-09-06:
Hi,

We are trying to get X working on a Lenovo L15 system which has AMD Ryzen 5 Pro 4650U with Radeon Graphics. 
The kernel version is 4.19
Output of lspci is;
VGA Compatible controller. Advanced Micro Devices inc. (AMD/ATI) Renoir (rev d3)

We do not need all the functionality, just basic graphics will be fine. 

Xorg log file; 

[Xorg.0.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/5180208/Xorg.0.log)

Regards,

Aydin

jinjianrong, 2020-09-07:
@paykocaydin This is not related with AMDVLK. You may need to upgrade kernel and ddx driver to newer version

JacobHeAMD, 2020-09-07:
"open /dev/dri/card0: No such file or directory" indicates that the kernel mode driver is not loaded, you can check it with "lsmod | grep amdgpu"? If it's not loaded, you can check what's going on with "dmesg"?

paykocaydin, 2020-09-07:
Hi,

lsmod | grep amdgpu produced nothing. So it is not loaded.
dmesg also shows nothing about amdgpu, not even an error message.

As far as i know the kernel that i am working on (4.19) does not support
this GPU.
I have learned that kernels from 5.3 have support.

So does this mean that the driver have to be backported to this kernel in
order to work.

If i am correct i can use framebuffer device to use graphics even my video
card is not supported.
Is that correct? Kernel has framebuffer enabled.

If yes can you please point me to a document on how to do this.
I do not need all the features, just basic X will be enough.

Many thanks for your time.

Aydin

closed 2021-03-05 by jinjianrong