GPUOpen-Drivers/AMDVLK #185: [DXVK] Visual artefacts in Assassins Creed: Odyssey

Brisse89, 2020-09-17:
Visual artefacts in Assassins Creed: Odyssey using DXVK 1.7.1 and AMDVLK 2020.Q3.5:

GPU: R9 Fury

There is no issue when using the mesa radv driver, only when using amdvlk.

[image]

Brisse89, 2020-10-20:
Not sure if this is any help, but I might as well mention it. I just upgraded to AMDVLK 2020.Q4.1 and the artefacts look different now. Those shown in the screenshot above are gone, and instead there are bright flashing colours.

JaxLinAMD, 2020-10-22:
thanks for the update

Brisse89, 2021-01-08:
Tested new release (AMDVLK 2021.Q1.1) and it now renders correctly, at least on my machine.

This issue can probably be closed now, but I'll leave it to you AMD-folks in case you want to do your own testing first.

closed 2021-03-09 by jinjianrong