GPUOpen-Drivers/AMDVLK #186: amdvlk segfaults when compiled without "debug"

justxi, 2020-09-20:
Hello,

when I build "amdvlk" without "debug" running "vulkaninfo" segfaults.

I use:
- Gentoo Linux 
- GCC 10.2.0
- Radeon RX 560

I tried the following versions:
- 2020-3.1
- 2020-3.2
- 2020-3.3
- 2020-3.5

The same problem with all versions.
Any hints how to "debug" this?

Flakebi, 2020-09-23:
Hi,

Can you try to compile with debug information? Does an assert fail then? Or does it segfault too?

Note that options like `-O3` and `-march=native` are not supported and are known to cause segfaults, so make sure to remove them from your compilation flags.

justxi, 2020-09-25:
Ok, replacing "-march=native" with "-march=x86-64" seems to solve the problem.
I can run "vulkaninfo" and "vkcube".

I will recompile with "debug" and check for "asserts".

justxi, 2020-09-25:
When compiling with "debug" I get the following asserts:
```
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_create_sem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_signal_sem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_wait_sem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_export_sem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_import_sem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_destroy_sem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_create_bo_from_phys_mem (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_bo_remap_secure (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_query_private_aperture (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_query_shared_aperture (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_bo_get_phys_address (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_reserved_vmid (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_unreserved_vmid (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_ctx_create3 (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/inc/util/palLibrary.h:84:GetFunction)
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/src/core/os/amdgpu/amdgpuDevice.cpp:671:EarlyInit)
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/pal/src/core/os/amdgpu/amdgpuDevice.cpp:671:EarlyInit)
```

Flakebi, 2020-09-25:
> Ok, replacing "-march=native" with "-march=x86-64" seems to solve the problem.

Nice.

The messages you see are only warnings, so they are fine.

justxi, 2020-09-25:
Thanks for the hint.

It it known what causes the segfault when using "-march=native"?

Flakebi, 2020-10-01:
I don’t think the cause is known. It would be nice to find and fix it though. Maybe compiling with -march=native and debug information can give a hint where the problem is.

justxi, 2020-10-04:
The problem is when compiling with "debug" running "vulkaninfo" and "vkcube" seem to work.

I wrote a small script to compare the option changed when using "march=native":
```
#!/bin/bash
gcc -march=x86-64 -Q --help=target > x86-64.txt
gcc -march=native -Q --help=target > native.txt
diff -Naur x86-64.txt native.txt > cmp.txt
grep "+ " cmp.txt
```

The result is:
```
+  -mabm                       		[enabled]
+  -madx                       		[enabled]
+  -maes                       		[enabled]
+  -march=                     		skylake
+  -mavx                       		[enabled]
+  -mavx2                      		[enabled]
+  -mbmi                       		[enabled]
+  -mbmi2                      		[enabled]
+  -mclflushopt                		[enabled]
+  -mcx16                      		[enabled]
+  -mf16c                      		[enabled]
+  -mfma                       		[enabled]
+  -mfsgsbase                  		[enabled]
+  -mfxsr                      		[enabled]
+  -mlzcnt                     		[enabled]
+  -mmmx                       		[enabled]
+  -mmovbe                     		[enabled]
+  -mno-sse4                   		[disabled]
+  -mpclmul                    		[enabled]
+  -mpopcnt                    		[enabled]
+  -mprfchw                    		[enabled]
+  -mrdrnd                     		[enabled]
+  -mrdseed                    		[enabled]
+  -msahf                      		[enabled]
+  -msgx                       		[enabled]
+  -msse                       		[enabled]
+  -msse2                      		[enabled]
+  -msse3                      		[enabled]
+  -msse4                      		[enabled]
+  -msse4.1                    		[enabled]
+  -msse4.2                    		[enabled]
+  -mssse3                     		[enabled]
+  -mtune=                     		skylake
+  -mxsave                     		[enabled]
+  -mxsavec                    		[enabled]
+  -mxsaveopt                  		[enabled]
+  -mxsaves                    		[enabled]
```
It seems that most of the enabled options are SIMD extensions.
The only I way I see to find options which results to the segfault, is to enable them step by step and see if the segfault occurs again.

Flakebi, 2020-10-04:
I think you should be able to compile in release mode with `-march=native -g`. That should produce the same code as `-march=native` but with debug symbols enabled, so we can look at where the segfault occurs.

justxi, 2020-10-04:
Ok, I will do that.

In the meantime I observed something interesting.
I use "VK_ICD_FILENAMES" to select the desired Vulkan implementation, e.g. I have RADV and Intel installed also.
When I run "> VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd64.json:/usr/share/vulkan/icd.d/intel_icd.x86_64.json vulkaninfo" with "march=x86-64" and all the GCC option enabled except "-mtune=skylake", I get a segfault at the "Device Group" info:

```
[..]
Device Groups:
==============
Group 0:
	Properties:
		physicalDevices: count = 1
			Intel(R) UHD Graphics 630 (CFL GT2) (ID: 0)
		subsetAllocation = 0

	Present Capabilities:
		Intel(R) UHD Graphics 630 (CFL GT2) (ID: 0):
			Can present images from the following devices: count = 1
				Intel(R) UHD Graphics 630 (CFL GT2) (ID: 0)
		Present modes: count = 1
			DEVICE_GROUP_PRESENT_MODE_LOCAL_BIT_KHR

Group 1:
	Properties:
		physicalDevices: count = 1
			Radeon RX 560 Series (ID: 0)
		subsetAllocation = 0

Segmentation fault (core dumped)
```

justxi, 2020-10-04:
So first I had to find out that "amdvlk" strip by default: 
https://github.com/GPUOpen-Drivers/xgl/blob/c6c90450fb1abf5f414acf1e38a0f51a72c426c1/CMakeLists.txt#L243

And now I have "vulkaninfo", "libvulkan" and "amvlk" with debug symbols.
Running "VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd64.json gdb vulkaninfo" gives me:
```
Reading symbols from vulkaninfo...
(gdb) run
Starting program: /usr/bin/vulkaninfo 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff2deb616 in vk::Device::Create(vk::PhysicalDevice*, VkDeviceCreateInfo const*, VkAllocationCallbacks const*, vk::DispatchableDevice**) [clone .isra.0] (pPhysicalDevice=0x555555a5ec28, 
    pCreateInfo=0x7fffffffc4b0, ppDevice=0x555555c17f00, pAllocator=<optimized out>) at /var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/xgl/icd/api/vk_device.cpp:381
381	/var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/xgl/icd/api/vk_device.cpp: No such file or directory.
(gdb) bt
#0  0x00007ffff2deb616 in vk::Device::Create(vk::PhysicalDevice*, VkDeviceCreateInfo const*, VkAllocationCallbacks const*, vk::DispatchableDevice**) [clone .isra.0] (pPhysicalDevice=0x555555a5ec28, 
    pCreateInfo=0x7fffffffc4b0, ppDevice=0x555555c17f00, pAllocator=<optimized out>) at /var/tmp/portage/media-libs/amdvlk-2020.3.5-r1/work/xgl/icd/api/vk_device.cpp:381
#1  0x00007ffff60a0ef9 in terminator_CreateDevice (physicalDevice=0x555555bceef0, pCreateInfo=<optimized out>, pAllocator=0x0, pDevice=0x7fffffffc908)
    at /var/tmp/portage/media-libs/vulkan-loader-1.2.141-r1/work/Vulkan-Loader-1.2.141/loader/loader.c:6812
#2  0x00007ffff60aafb8 in loader_create_device_chain (pd=pd@entry=0x555555bceef0, pCreateInfo=pCreateInfo@entry=0x7fffffffcbc0, pAllocator=pAllocator@entry=0x0, inst=inst@entry=0x5555557ed0e0, 
    dev=dev@entry=0x555555c16d50, callingLayer=callingLayer@entry=0x0, layerNextGDPA=0x0) at /var/tmp/portage/media-libs/vulkan-loader-1.2.141-r1/work/Vulkan-Loader-1.2.141/loader/loader.c:6174
#3  0x00007ffff60abb8a in loader_layer_create_device (instance=instance@entry=0x0, physicalDevice=physicalDevice@entry=0x555555bcede0, pCreateInfo=pCreateInfo@entry=0x7fffffffcbc0, 
    pAllocator=pAllocator@entry=0x0, pDevice=pDevice@entry=0x555555c07778, layerGIPA=layerGIPA@entry=0x0, nextGDPA=0x0)
    at /var/tmp/portage/media-libs/vulkan-loader-1.2.141-r1/work/Vulkan-Loader-1.2.141/loader/loader.c:5783
#4  0x00007ffff60ae8af in vkCreateDevice (physicalDevice=0x555555bcede0, pCreateInfo=0x7fffffffcbc0, pAllocator=0x0, pDevice=0x555555c07778)
    at /var/tmp/portage/media-libs/vulkan-loader-1.2.141-r1/work/Vulkan-Loader-1.2.141/loader/trampoline.c:779
#5  0x00005555555b4020 in AppGpu::AppGpu (this=0x555555c06a70, inst=..., id=<optimized out>, phys_device=<optimized out>, chainInfos=...)
    at /var/tmp/portage/dev-util/vulkan-tools-1.2.141/work/Vulkan-Tools-1.2.141/vulkaninfo/vulkaninfo.h:1429
#6  0x0000555555564b5a in main (argc=<optimized out>, argv=<optimized out>) at /var/tmp/portage/dev-util/vulkan-tools-1.2.141/work/Vulkan-Tools-1.2.141/vulkaninfo/vulkaninfo.cpp:825

```

justxi, 2020-10-04:
That should be this one: https://github.com/GPUOpen-Drivers/xgl/blob/6e4aaa87a128f91cc62d337642949ee8cfcaeec8/icd/api/vk_device.cpp#L381

justxi, 2020-10-04:
After disabling "avx" and "avx2" it does not seem to "segfault" with "march=native".

justxi, 2020-10-04:
Maybe I hit the problem, which was discussed earlier here -> https://github.com/GPUOpen-Drivers/AMDVLK/issues/50 ?

justxi, 2020-10-04:
> Note that options like `-O3` and `-march=native` are not supported and are known to cause segfaults, so make sure to remove them from your compilation flags.

I noticed while compiling "amdvlk" that there is the "-O3"  option set. Due to the fact that I only set "-O2" on my system this must be in the build scripts.

Flakebi, 2020-10-05:
Thanks for the debugging, it is helpful. On Ryzen, `-march=native` does not include avx, so I needed to explicitly pass `-mavx -mavx2`.

The reason for the segfault with avx enabled is that some avx memory instructions expect aligned memory addresses, sometimes the stack is not properly aligned.
The failing instructions (e.g. `vmovdqa`) need an alignment of 32 Byte. We compile with `-mpreferred-stack-boundary=6`, so gcc expects the stack to be aligned to 64 Byte (lower 6 bits are zero). However, the driver does not setup the stack and applications usually have a lower alignment requirement of 16 Byte, so the stack is actually only aligned to 16 Byte and avx instructions will fail. For reference, there has been some discussion in gcc for similar bugs, e.g. in https://gcc.gnu.org/bugzilla/show_bug.cgi?id=40838.

To solve this problem, gcc needs to be instructed to realign the stack when necessary. Passing `-mincoming-stack-boundary=4` will insert stack realignment code into every function, this works but potentially slows down the driver unnecessarily. I think a better way is to add `__attribute__((force_align_arg_pointer))` to all entry functions of the driver.

I tried adding the attribute by setting `#define VKAPI_ATTR __attribute__((force_align_arg_pointer))` in `vk_platform.h`, this adds code to realign the stack to 64 Byte and at least `vulkaninfo` and `vkcube` are not crashing anymore.

justxi, 2020-10-05:
@Flakebi Have you checked in the changes? 
Then I´m looking forward to test it.

Flakebi, 2020-10-05:
I still need to figure out a sane way to add these attributes to the functions.

justxi, 2020-10-07:
If you have something to test, please drop a note here. 
Thanks.

justxi, 2020-11-08:
Is there any progress on this?

staalmannen, 2020-11-09:
I am not entirely sure if I have encountered the same issue with amdvlk on top of amdgpu drivers in Arch linux.

vulkaninfo crashes with a core dump on Q4.2 and Q4.3, but works on Q4.1
I also tried amdvlk-git from AUR (v.2020.Q4.2.r1.a69cae2-1) and it also crashes with a core dump.

```
lspci -v | grep VGA 
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Mullins [Radeon R4/R5 Graphics] (rev 45) (prog-if 00 [VGA controller])
```

Flakebi, 2020-11-09:
@staalmannen I just tried the Q4.3 driver from the arch repos and it didn’t crash for me when running vulkaninfo or vkcube.
I’d guess this is an unrelated crash but it should be easy to verify. If you run with a debugger attached and the instruction where the segfault occurs (or the one before) requires aligned memory, you probably hit this issue.

@justxi Sorry, no progress. I pushed a fix here: https://github.com/GPUOpen-Drivers/xgl/compare/dev...Flakebi:fix-stack-alignment, unfortunately it can’t go upstream in this form because the attribute does not belong to the vulkan headers.
If someone shows that compiling with -march=native makes the driver faster, it might be fixed sooner :)

staalmannen, 2020-11-10:
@Flakebi Indeed it seems to be a separate issue. Should I open another issue here at github?

`(gdb) run
Starting program: /usr/bin/vulkaninfo
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
ERROR: [Loader Message] Code 0 : /usr/lib32/amdvlk32.so: wrong ELF class: ELFCLASS32

Program received signal SIGABRT, Aborted.
0x00007ffff7a50615 in raise () from /usr/lib/libc.so.6`

Flakebi, 2020-11-10:
Yes, please open another issue.
Can you please write down your OS, driver version and hardware there? (as you did in the comment here)
Can you also run `env VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd64.json vulkaninfo` to make sure no other driver is loaded and print the backtrace in gdb (`bt`)?

Btw, you can use triple backticks for format code blocks in markdown (` ``` `).

closed 2021-03-09 by jinjianrong

Flakebi, 2021-07-09:
This issue is at last fixed in 2021.Q3.1. Compiling with avx and avx2 support is working now.

justxi, 2021-07-09:
Nice. I am intessted how it was fixed, can you point to the location in the source code?

Flakebi, 2021-07-09:
It’s the diff here, the `-mpreferred-stack-boundary` flag was removed: https://github.com/GPUOpen-Drivers/pal/commit/ad699adac6f9f331bbc454050f6b40d1549ce752#diff-0287c242e7b446e5aec52b4b5c48e24bdb78626e34079843ca768ab6bf1aefff