GPUOpen-Drivers/AMDVLK #187: AMDVLK and vulkan-amdgpu-pro give out of memory errors when an AMD GPU isn't installed

gardotd426, 2020-09-26:
I know the title might sound confusing, but I'll try to explain it so it makes sense...

So I recently went from an AMD to an Nvidia GPU on this machine and put my AMD GPU in another machine. Anyway, I had amdvlk and vulkan-amdgpu-pro installed, along with RADV and vulkan-intel. So I had 5 vulkan icd files in /usr/share/vulkan/icd.d. Before back when I had the AMD GPU, any vulkan programs or `vulkaninfo` would just skip over any incompatible or irrelevant icd files. For example, I didn't have any Intel graphics, so it would just ignore the intel icds. But after removing the AMD GPU, I get 
```
ERROR at /build/vulkan-tools/src/Vulkan-Tools-1.2.153/vulkaninfo/vulkaninfo.h:667:vkCreateInstance failed with ERROR_OUT_OF_HOST_MEMORY
```
Instead of just skipping past the amdvlk/vulkan-amdgpu-pro icd files, an error is given. This breaks Vulkan support for all programs (unless you set `VK_ICD_FILENAMES` with a specific file). AMDVLK and vulkan-amdgpu-pro are the only drivers that do this. vulkan-radeon, vulkan-intel, and the proprietary Nvidia driver do as they're supposed to and just don't activate if there's no compatible GPU installed. I was told by a wine dev that this out of memory error is not supposed to happen, and to report the issue here. 

Here's the output of `VK_LOADER_DEBUG=all vulkaninfo` with amdvlk installed. 

[vkloaderdebug-amdvlk.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/5287455/vkloaderdebug-amdvlk.txt)

Final note, obviously upon removing amdvlk and vulkan-amdgpu-pro everything works as intended, despite also having vulkan-radeon and vulkan-intel installed.

qiaojbao, 2021-03-23:
Can be reproduced with AMDVLK, will fix it soon.

jinjianrong, 2021-04-28:
Fixed in 2021-Q2.2

closed 2021-04-28 by jinjianrong