GPUOpen-Drivers/AMDVLK #188: "amdgpu_device_initialize: DRM version is 2.50.0" with two videocards: radeon+amdgpu

jl452, 2020-10-05:
i have 2 videocards: radeon+amdgpu
and have error:
VK_ICD_FILENAMES=/etc/vulkan/icd.d/amd_icd64.json vulkaninfo 
amdgpu_device_initialize: DRM version is 2.50.0 but this driver is only compatible with 3.x.x.
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
/build/vulkan-tools-KEbD_A/vulkan-tools-1.2.131.1+dfsg1/vulkaninfo/vulkaninfo.h:371: failed with ERROR_INCOMPATIBLE_DRIVER

but i have vulkan with mesa:
vulkaninfo
...
VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.1.73, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 6
                GPU id  : 0 (AMD RADV PITCAIRN (ACO))
                Layer-Device Extensions: count = 0
                GPU id  : 1 (llvmpipe (LLVM 10.0.1, 256 bits))
                Layer-Device Extensions: count = 0
                GPU id  : 2 (llvmpipe (LLVM 10.0.1, 256 bits))
                Layer-Device Extensions: count = 0
                GPU id  : 3 (AMD RADV PITCAIRN (ACO))
                Layer-Device Extensions: count = 0
                GPU id  : 4 (llvmpipe (LLVM 10.0.1, 256 bits))
                Layer-Device Extensions: count = 0
                GPU id  : 5 (AMD RADV PITCAIRN (ACO))
                Layer-Device Extensions: count = 0
...
VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.1.73, layer version 1:
...
VkPhysicalDeviceDriverProperties:
---------------------------------
        driverID           = DRIVER_ID_MESA_RADV
        driverName         = radv
        driverInfo         = Mesa 20.3.0-devel (git-728061b 2020-09-30 focal-oibaf-ppa) (ACO)
        conformanceVersion = 1.2.3.0
...

dpkg -S /etc/vulkan/icd.d/amd_icd64.json
amdvlk:amd64: /etc/vulkan/icd.d/amd_icd64.json

apt policy amdvlk
amdvlk:
  Installed: 2020.Q3.6
  Candidate: 2020.Q3.6
  Version table:
 *** 2020.Q3.6 500
        500 http://repo.radeon.com/amdvlk/apt/debian bionic/main amd64 Packages

hwinfo --gfxcard
1)
  Model: "ATI Richland [Radeon HD 8650G]"
  Vendor: pci 0x1002 "ATI Technologies Inc"
  Device: pci 0x990b "Richland [Radeon HD 8650G]"
  SubVendor: pci 0x1462 "Micro-Star International Co., Ltd. [MSI]"
  SubDevice: pci 0x10ef 
  Driver: "radeon"
  Driver Modules: "radeon"
2)
  Model: "Micro-Star International Co., Ltd. [MSI] Radeon R9 M290X"
  Vendor: pci 0x1002 "ATI Technologies Inc"
  Device: pci 0x6801 "Neptune XT [Radeon HD 8970M]"
  SubVendor: pci 0x1462 "Micro-Star International Co., Ltd. [MSI]"
  SubDevice: pci 0x1117 "Radeon R9 M290X"
  Driver: "amdgpu"
  Driver Modules: "amdgpu"

uname -a
Linux ... 5.8.12-050812-lowlatency #202009261732 SMP PREEMPT Sat Sep 26 17:38:20 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

ubuntu 20.04

qiaojbao, 2020-10-14:
This limitation is from libdrm

amdgpu_device.c, drm_public int amdgpu_device_initialize()
                version = drmGetVersion(fd);
                if (version->version_major != 3) {
                                fprintf(stderr, "%s: DRM version is %d.%d.%d but this driver is "
                                                "only compatible with 3.x.x.\n",
                                                __func__,
                                                version->version_major,
                                                version->version_minor,
                                                version->version_patchlevel);
                                drmFreeVersion(version);
                                r = -EBADF;
                                goto cleanup;
                }

jl452, 2020-10-14:
can it be fixed?
need i create new issue in other project?

qiaojbao, 2020-10-14:
AMDVLK driver base on new amdgpu driver module.
For this vulkaninfo drm warning issue with AMDVLK, we will fix it.

qiaojbao, 2020-10-28:
Fixed in internal.

jinjianrong, 2020-11-23:
@jl452 please check if the issue is fixed with Q4.4 release

closed 2020-11-23 by jinjianrong

jl452, 2020-11-23:
@jinjianrong fixed, thanks

```
VK_ICD_FILENAMES=/etc/vulkan/icd.d/amd_icd64.json vulkaninfo
...
        Device Group Properties (Group 0):
        ----------------------------------
                physicalDeviceCount: count = 1
                        AMD Radeon(TM) HD8970M (ID: 0)
                subsetAllocation = 0
        Device Group Present Capabilities (Group 0):
                AMD Radeon(TM) HD8970M (ID: 0)
                Can present images from the following devices:
                        AMD Radeon(TM) HD8970M (ID: 0)
```
```
dpkg -S /etc/vulkan/icd.d/amd_icd64.json
amdvlk:amd64
 *** 2020.Q4.5 500
        500 http://repo.radeon.com/amdvlk/apt/debian bionic/main amd64 Packages
```