GPUOpen-Drivers/AMDVLK #189: incorrect image

JohnnyDilinjer, 2020-11-04:
Hello. When using AMDVLK, such a picture appears, when using RADV, the picture is correct.
How can I find the reason?
[image]

qiaojbao, 2020-11-05:
I can not reproduce it with Q4_2 AMDVLK on ubuntu 20.04.
Could you provide more info about your system and cards info?
Guess something wrong with Xserver (maybe others) framebuffer format.

JohnnyDilinjer, 2020-11-06:
I have a not quite standard system. I just wonder how can I find the cause of this wrong picture?
[image]

qiaojbao, 2020-11-06:
Maybe we can start at demo_create_surface and vulkan swap-chain.

closed 2021-03-10 by jinjianrong