GPUOpen-Drivers/AMDVLK #192: Many DXVK games display transparent artifacts all over the screen when using AMDVLK

niobium93, 2020-11-29:
Specs:
Arch Linux
AMD Radeon Rx 5700 XT
Kernel: 5.9.10-zen
AMDVLK 2020.Q4.5
DXVK v1.7.2-26-ge79e8b90
wine-tkg-staging-fsync-git-5.22.r5.g991f2e8e-309
[image]
[image]
[image]
[image]
[image]
Games in order: Myst Online Uru Live, Call of Duty Modern Warfare 2, Horizon Chase Turbo, Call of Duty 4, Trackmania United Forever.
Most of these are D3D9 games with the exception of Horizon Chase Turbo which uses D3D11. Many other D3D11 games work fine though.

aufkrawall, 2021-01-07:
Does the issue occur only with MSAA? I'm seeing a similar issue with MSAA across several games on Navi 10, e.g. Mirrors' Edge (2009 Unreal Engine 3). This wasn't broken on Polaris.

niobium93, 2021-01-07:
[Yep! That's what's causing it.](https://www.youtube.com/watch?v=r2NatAtkIgM)
Tried with Trackmania and COD4 too. It's only broken with MSAA.

aufkrawall, 2021-01-08:
Issue is fixed here with v-2021.Q1.1.

niobium93, 2021-01-08:
Yep, fixed with 2021.Q1.1.

closed 2021-01-08 by niobium93