GPUOpen-Drivers/AMDVLK #194: ShaderCacheMode,2 lets amdgpu-pro Vulkan driver write shader cache to disk twice

aufkrawall, 2021-01-07:
amdgpu-pro Vulkan driver has on-disk shader cache enabled by default for some time now, it stores them at `~/.cache/AMD/VkCache`. But when you set `ShaderCacheMode,2` in `amdPalSettings.cfg`, it additionally also creates `~/.cache/AMD/ScpcCache` and stores cached shaders there as well. This is redundant and one of the paths shouldn't be used.

At this occasion, it would be nice to enable the on-disk shader cache by default also for the amdvlk-open driver to get it in line with the -pro/Windows driver. Especially since the shader compile times with the open compiler take longer.

Kai-AMD, 2021-01-13:
Hi @aufkrawall,

As far as I know, amdgpu-pro Vulkan driver has two cache mechanisms, one is pipeline cache and the other one is shader cache. So as you mentioned, ~/.cache/AMD/VkCache is for pipeline cache and ~/.cache/AMD/ScpcCache is for shader cache.
For amdvlk-open driver, the pipeline cache has been enabled by default.

Thanks,
Kai

aufkrawall, 2021-01-13:
Is there any difference in practice between pipeline and and shader cache in amdvlk? With amdvlk-pro, shader compile stutter is definitely eliminated by the data stored by default in `VkCache`. amdvlk-open doesn't store any data by default in `~/.cache/AMD`. It only does so when explicitly setting `ShaderCacheMode,2`. Without this option set, games played with amdvlk-open show bad loading times and shader compile stutter every time they are restarted, as shader cache is only stored in RAM (also PAL docs say that). This is different with amdvlk-pro.

aufkrawall, 2021-01-13:
Ok, I stand corrected: Like you said, also recent amdvlk-open releases indeed cache shaders inside `VkCache` by default and pipeline caches stored in `LlpcCache` and `ScpcCache` are indeed just tiny. In fact, they seem to be just empty 0 byte files with amdvlk-pro.

Thanks for turning shader cache on by default!

closed 2021-01-13 by aufkrawall