GPUOpen-Drivers/AMDVLK #195: multiple issues with VK_LAYER_AMD_switchable_graphics

DadSchoorse, 2021-01-08:
Hi, I noticed multiple technical issues with the new layer:

- it doesn't handle multiple VkInstances correctly: [g_nextLinkFuncs](https://github.com/GPUOpen-Drivers/xgl/blob/3237b046add2809634e7f5f3fcdea1af4ad6613d/icd/layers/vk_layer_switchable_graphics.cpp#L41) should store function pointers per instance because they aren't guaranteed to be identical across them.
- the 2 jsons (for i386 and amd64) have the same [name](https://github.com/GPUOpen-Drivers/AMDVLK/blob/0b8a7192d4d1a121757b8ba0d5f7692bc04991d2/json/Ubuntu/amd_icd64.json#L9). The Vulkan loader will discard one  because of that. (See https://github.com/KhronosGroup/Vulkan-Loader/issues/155 and https://github.com/KhronosGroup/Vulkan-Loader/pull/525)
- a smaller issue (because many layers misbehave similar): `vk_icdGetInstanceProcAddrSG(NULL, "vkGetInstanceProcAddr")` returns the next function in the chain instead of itself. 

Also, I would like to ask why VK_LAYER_AMD_switchable_graphics exists to begin with. It's already possible to choose a driver with `VK_ICD_FILENAMES`. If the intended effect is to have distro install the layer alongside amdvlk it will cause a lot of confusion why radv doesn't work anymore.

soararing, 2021-01-11:
The reason to introduce VK_LAYER_AMD_switchable_graphics in Linux: Ubuntu 20.04 installs RADV by default, after install AMDVLK, there are two Vulkan ICDes installed, for each Vulkan app, vkEnumeratePhysicalDevices will return two physical devices,  Vulkan apps don't know which physical device should be used, then we need a mechanism to identify which physical device should be used by Vulkan apps. With the AMD switchable graphics layer support, when both RADV and AMDVLK are installed, AMDVLK will be used by default, only AMDVLK physical device is returned to Vulkan apps, but at this time, if Vulkan apps wants to use RADV, end users can specify environment variable AMD_VULKAN_ICD = RADV to switch from AMDVLK to RADV, and use AMD_VULKAN_ICD = AMDVLK to switch back to AMDVLK, basically the layer has the same functionality with VK_ICD_FILENAMES, but AMD_VULKAN_ICD is much easier to use than VK_ICD_FILENAMES, especially in the case of Intel iGPU + AMD dGPU, without the layer, we may need to specify both Intel and AMD JSON file path to VK_ICD_FILENAMES.

soararing, 2021-01-11:
>>>it doesn't handle multiple VkInstances correctly: g_nextLinkFuncs should store function pointers per instance because they aren't guaranteed to be identical across them.

I don't see it is a problem, g_nextLinkFuncs is a global variable in the layer, there are only several Vulkan ICD function pointers stored in it, for different VkInstances in the same app process, I think these ICD function pointers have the same value.

soararing, 2021-01-11:
>>>a smaller issue (because many layers misbehave similar): vk_icdGetInstanceProcAddrSG(NULL, "vkGetInstanceProcAddr") returns the next function in the chain instead of itself.

I will check the side effect of this behavior, we may need to fail the call if a NULL VkInstance is input.

DadSchoorse, 2021-01-11:
> for different VkInstances in the same app process, I think these ICD function pointers have the same value for different VkInstances.

The scenario I see is: The application creates 2 instances, with different layers.
instance 1 has this chain:
- Vulkan loader
- VK_LAYER_AMD_switchable_graphics
- <other layer which intercepts functions used by VK_LAYER_AMD_switchable_graphics>
- loader terminators and ICDs 

instance 2:
- Vulkan loader
- VK_LAYER_AMD_switchable_graphics
- loader terminators and ICDs 

with the creation of instance 2 the global function pointers in VK_LAYER_AMD_switchable_graphics will be overwritten. Now the application calls one of the functions that are implemented by VK_LAYER_AMD_switchable_graphics with instance 1. Instead of calling the function in the other layer VK_LAYER_AMD_switchable_graphics will directly call the terminators which is not what should happen.

reactormonk, 2021-01-11:
Vulkan straight up broke here because of it: https://bbs.archlinux.org/viewtopic.php?pid=1949414

soararing, 2021-01-12:
Thanks @DadSchoorse  & @reactormonk, I will try to verify the issues.

Joshua-Ashton, 2021-01-12:
This layer breaks vulkaninfo... :l

Joshua-Ashton, 2021-01-12:
Please disable this layer until you at-least fix it. This is completely unusable and it breaks multiple apps (including vulkaninfo) right now.

Flakebi, 2021-01-12:
See also #196 for ERROR_INITIALIZATION_FAILED.

@Joshua-Ashton: Removing `/usr/share/vulkan/implicit_layer.d/amd_icd64.json` disables the layer.

Joshua-Ashton, 2021-01-12:
I know how to work around it. it, I just don't understand how you can release something so completely broken and not immediately at least disable what is causing the problem for users.

DiabeticCrab, 2021-01-12:
I would really like to see VK_LAYER_AMD_switchable_graphics getting removed and a more standard approach to be taken - i.e. like working with Khronos to make additions to vulkan-loader.

jinjianrong, 2021-01-13:
The 2021.Q1.1  package is updated with the AMD_switchable_graphics  layer disabled. The issue will be fixed in next release. Thanks All.

jinjianrong, 2021-04-08:
The issue is fixed in 2021.Q2.1 release.

closed 2021-04-08 by jinjianrong