GPUOpen-Drivers/AMDVLK #196: 2021.Q1.1-1 yields ERROR_INITIALIZATION_FAILED

milianw, 2021-01-10:
I'm on archlinux with linux 5.10.5-arch1-1, mesa 20.3.2-2. The recent update to amdvlk 2021.Q1.1-1 breaks vulkan for me on a AMD Navi 10 GPU (5700 XT):

```
$ vulkaninfo
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.164

Instance Extensions: count = 18
===============================
        VK_EXT_acquire_xlib_display            : extension revision 1
        VK_EXT_debug_report                    : extension revision 9
        VK_EXT_debug_utils                     : extension revision 2
        VK_EXT_direct_mode_display             : extension revision 1
        VK_EXT_display_surface_counter         : extension revision 1
        VK_KHR_device_group_creation           : extension revision 1
        VK_KHR_display                         : extension revision 23
        VK_KHR_external_fence_capabilities     : extension revision 1
        VK_KHR_external_memory_capabilities    : extension revision 1
        VK_KHR_external_semaphore_capabilities : extension revision 1
        VK_KHR_get_display_properties2         : extension revision 1
        VK_KHR_get_physical_device_properties2 : extension revision 2
        VK_KHR_get_surface_capabilities2       : extension revision 1
        VK_KHR_surface                         : extension revision 25
        VK_KHR_surface_protected_capabilities  : extension revision 1
        VK_KHR_wayland_surface                 : extension revision 6
        VK_KHR_xcb_surface                     : extension revision 6
        VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 9
=================
VK_LAYER_AMD_switchable_graphics (AMD switchable graphics layer) Vulkan version 1.2.164, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id = 0 (AMD Radeon RX 5700 XT)
                Layer-Device Extensions: count = 0

VK_LAYER_AMD_switchable_graphics (AMD switchable graphics layer) Vulkan version 1.2.164, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id = 0 (AMD Radeon RX 5700 XT)
                Layer-Device Extensions: count = 0

VK_LAYER_KHRONOS_validation (Khronos Validation Layer) Vulkan version 1.2.162, layer version 1:
        Layer Extensions: count = 3
                VK_EXT_debug_report        : extension revision 9
                VK_EXT_debug_utils         : extension revision 1
                VK_EXT_validation_features : extension revision 2
        Devices: count = 1
                GPU id = 0 (AMD Radeon RX 5700 XT)
                Layer-Device Extensions: count = 3
                        VK_EXT_debug_marker     : extension revision 4
                        VK_EXT_tooling_info     : extension revision 1
                        VK_EXT_validation_cache : extension revision 1

VK_LAYER_LUNARG_api_dump (LunarG API dump layer) Vulkan version 1.2.154, layer version 2:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id = 0 (AMD Radeon RX 5700 XT)
                Layer-Device Extensions: count = 1
                        VK_EXT_tooling_info : extension revision 1

VK_LAYER_LUNARG_device_simulation (LunarG device simulation layer) Vulkan version 1.2.154, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id = 0 (AMD Radeon RX 5700 XT)
                Layer-Device Extensions: count = 1
                        VK_EXT_tooling_info : extension revision 1
[... 2536 lines cut ...]
	runtimeDescriptorArray                             = true
	samplerFilterMinmax                                = true
	scalarBlockLayout                                  = true
	imagelessFramebuffer                               = true
	uniformBufferStandardLayout                        = true
	shaderSubgroupExtendedTypes                        = true
	separateDepthStencilLayouts                        = true
	hostQueryReset                                     = true
	timelineSemaphore                                  = true
	bufferDeviceAddress                                = true
	bufferDeviceAddressCaptureReplay                   = true
	bufferDeviceAddressMultiDevice                     = true
	vulkanMemoryModel                                  = true
	vulkanMemoryModelDeviceScope                       = true
	vulkanMemoryModelAvailabilityVisibilityChains      = false
	shaderOutputViewportIndex                          = true
	shaderOutputLayer                                  = true
	subgroupBroadcastDynamicId                         = true

VkPhysicalDeviceVulkanMemoryModelFeatures:
------------------------------------------
	vulkanMemoryModel                             = true
	vulkanMemoryModelDeviceScope                  = true
	vulkanMemoryModelAvailabilityVisibilityChains = false
```

Flakebi, 2021-01-11:
Hi,
I don’t see this crash on Navi14 (RX 5300).
From your output, it seems like you have installed both, amdvlk and radv? 2021.Q1.1 introduced a Vulkan layer related to radv (VK_LAYER_AMD_switchable_graphics), so this might be related.
Can you try if (temporarily) uninstalling radv fixes the issue?

milianw, 2021-01-11:
indeed, removing `vulkan-radeon` fixes it for me - thank you! still, this should probably get fixed somehow, or at least the error messages be improved to make it more obvious. at least I hope people can find this thread now to find the (easy) workaround.

thank you!

soararing, 2021-01-12:
I will fix this ERROR_INITIALIZATION_FAILED layer issue soon.

pdaniell-nv, 2021-02-23:
@soararing has this fix been implemented yet? I'm seeing the same issue on Windows with the latest 21.2.3 driver here: https://github.com/KhronosGroup/Vulkan-Loader/issues/552#issuecomment-784564575

jiansan, 2021-02-25:
Fix in AMDVLK is on the way to be released. Now VK_LAYER_AMD_switchable_graphics is disabled in AMDVLK. It will be enabled again together with the fix.

jinjianrong, 2021-04-08:
The issue is fixed in 2021.Q2.1 release

closed 2021-04-08 by jinjianrong