GPUOpen-Drivers/AMDVLK #199: amdgpu driver hangs and crashs

Darhal, 2021-02-06:


Darhal, 2021-02-09:
Its probably also important to add that even with gpu_recovery enabled (after the driver crashes due to a VK app running), on shutting down the PC, it hangs in a black screen that show these lines and I have to turn off the power to shut the computer down:
```
kernel: [ 1169.446678] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=124, emitted seq=126
kernel: [ 1169.446832] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process  pid 0 thread  pid 0
```

Flakebi, 2021-02-24:
Hi,
I think your GPU is too old to support Vulkan, at least I didn’t find any note that says it is supported. Searching for 8670M on the [driver download site](https://www.amd.com/en/support/graphics/amd-radeon-hd/amd-radeon-hd-8000m-series/amd-radeon-hd-8670m-series-gpu) lists OpenGL 4.4 and DirectX 11 as the latest supported APIs (and the latest driver is from end of 2015).
For completeness, if I go the version numbers right, Sun XT corresponds to gfx6.

That said, did you try if radv works? (The Vulkan driver that is part of mesa.) It should be registered at `/etc/vulkan/icd.d/radeon_icd64.json`.

Darhal, 2021-02-24:
Hello, thanks for the reply.

Actually my GPU is "AMD Radeon(TM) R5 M430 2GB DDR 3" (this is according to the laptop vendor). In the BIOS it shows R5 M430 as well. The following website shows my GPU as well: https://vulkan.gpuinfo.org/ so it does indeed support Vulkan.

I have tried radv driver, it shows a warning then crashes as well upon creating the device.

The 'amdvlk' driver actually renders few frames before crashing.

Flakebi, 2021-02-24:
Aha, thanks for the clarification. From what I heard internally, gfx6 is unfortunately not officially supported by amdvlk (the Linux driver, Windows seems to be supported). It is GCN 1.0, so it *could* work, but we’re not testing it.

If you want to debug that further, [umr](https://gitlab.freedesktop.org/tomstdenis/umr) is a good starting point. If you disable gpu recovery and ssh into the machine, it can dump some information.
`umr -t` gives an overview over which parts of the hardware are active (the ones that display 100 % are hanging).
`umr -O bits,disasm_anyways --waves gfx_0.0.0` shows the currently running shader and info like if the shader segfaulted.

Darhal, 2021-02-26:
`sudo umr -t` yields the following:
```
ERROR: Device 0x1916 not found in UMR device table
ASIC not found (instance=0, did=ffffffffffffffff)
```

Flakebi, 2021-03-02:
Hm, that makes debugging more difficult.
Does it hang for every Vulkan application? Can you try if `vulkaninfo` and `vkcube` are working? They should be part of the Vulkan SDK or a vulkan-tools package if there is one.

I’m not sure how to grasp the issue, my current ideas are
- if you can find a previous driver version that works, it can be bisected to a change that introduced the regression (though that assumes it is a regression)
- you can try to pinpoint it to some specific app behavior that introduce the hang, the [Conformance Test Suite](https://github.com/KhronosGroup/VK-GL-CTS/blob/master/external/vulkancts/README.md) is a good start as it tests many different features in isolation

Darhal, 2021-03-15:
Hello, sorry for the late response.

Sascha Willems examples and my engine makes the GPU hangs. I tried to run steam using vulkan, it crashed reporting that it can't find swapchain extensions)

I managed to get the [vkvia.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/6141935/vkvia.zip) and the [vulkaninfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/6141932/vulkaninfo.txt) files.
I have tried to run ```vkcube --gpu_number 1``` and it worked perfectly (0 being intel's iGPU and 1 the AMD one).

I still have to run the CTS, I will let you know when I get the results.

closed 2024-05-22 by jinjianrong