GPUOpen-Drivers/AMDVLK #200: Request - depth format feature SAMPLED_IMAGE_FILTER_LINEAR

LendoK, 2021-02-07:
On AMD RADV NAVI14 (ACO) there seems to be no depth format that supports format feature SAMPLED_IMAGE_FILTER_LINEAR.
Without SAMPLED_IMAGE_FILTER_LINEAR it is not possible to filter shadow maps for example.
Would be nice to have linear filtering for this formats:

```
VK_FORMAT_D32_SFLOAT_S8_UINT,
VK_FORMAT_D32_SFLOAT,
VK_FORMAT_D24_UNORM_S8_UINT,
VK_FORMAT_D16_UNORM_S8_UINT,
VK_FORMAT_D16_UNORM
```

Flakebi, 2021-02-08:
Hi, RADV and ACO are developed in the mesa repository: https://gitlab.freedesktop.org/mesa/mesa/-/issues
AMDVLK, which is developed in this repository, should show up as `AMD Radeon RX 5700 XT` or similar.

LendoK, 2021-02-08:
@Flakebi thanks for the info. Now I'm a little bit confused, why are there two open source drivers for the same card?

Flakebi, 2021-02-08:
That has historic reasons.
Back when Vulkan was first released, AMD announced that they will open source their Linux Vulkan driver. However, for a long time, only a closed source driver was available until at one point David Airlie (a mesa developer from RedHat) started his own Vulkan driver called ‘RADV’ as part of mesa (derived from **Rad**eon **V**ulkan). A while later, AMD finally open sourced their driver called ‘AMDVLK’. Since then, both of these drivers are actively developed. Some time later, RADV developers created their own compiler backend ACO (**A**MD **Co**mpiler) as an alternative to LLVM.

As radv is part of mesa (which contains most Linux graphics drivers), it is installed on most Linux systems. AMD provides Ubuntu and RHEL packages of AMDVLK and some distributions like Arch and NixOS provide their own packages for it.

So, there are 3 different Linux Vulkan drivers for AMD hardware, each sharing different parts of the code:
- AMDVLK: Shares code with the windows driver, uses LLVM as the shader compiler, developed by AMD
- amdgpu-pro: Closed source, same stack as AMDVLK, but uses the closed source compiler instead of LLVM, developed by AMD
- RADV+LLVM: Based on mesa, so shares code with OpenGL and other drivers, uses LLVM as the shader compiler, developed by RedHat, Valve and others
- RADV+ACO: Same, but uses ACO as the shader compiler, developed by RedHat, Valve and others

The kernel driver (called amdgpu) is shared between all of these drivers.

LendoK, 2021-02-08:
Ok, thanks again for this explanation. I hoped everything would get easier after moving from nvidia xD.
I installed this arch packages: mesa + vulkan-radeon. I'm not sure if AMDVLK is available on arch anymore

Flakebi, 2021-02-08:
I’d say it certainly is easier to run Wayland with a window manager like sway (which is currently not supported by the proprietary nvidia driver) :)
Arch is the only distribution I know that packages these parts of mesa separately. `vulkan-radeon` is radv and the `amdvlk` package is AMD’s official driver.
You can have both of them installed, they will be exposed as different “physical devices”. Also, it’s possible to force a driver by setting `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/<driver>.json`.

LendoK, 2021-02-08:
Linear filtering for depth formats works perfectly fine with `amdvlk`. I will stick to this package.
I don't understand why `amdvlk` package is not linked on this wiki page https://wiki.archlinux.org/index.php/AMDGPU

Flakebi, 2021-02-09:
Good to hear that it works with amdvlk. That is the advantage of having multiple drivers for the same thing I guess ;)
Fixed the wiki, the [Vulkan](https://wiki.archlinux.org/index.php/Vulkan) article goes more into the details btw.

closed 2021-03-09 by jinjianrong