GPUOpen-Drivers/AMDVLK #203: Overall Broken-ness with DXVK 1.8.1

Joshua-Ashton, 2021-03-03:
Hello! Here are a collection of bugs that have surfaced in the AMDVLK/AMDVLK-PRO driver following the release of DXVK 1.8.1

These games and their traces seem to replay fine on NVIDIA and with the RADV driver, with no validation errors but are seemingly broken under AMDVLK now.

We made some changes to the way we do image layouts in this release so that is probably related.

________

*Metal Gear Rising: Revengeance: visual artifacts when MSAA is enabled*: https://github.com/doitsujin/dxvk/issues/1961

*After updating to 1.8.1, edge smoothing in Sims 2 causes rainbow artifacts*: https://github.com/doitsujin/dxvk/issues/1957

*Skyrim Special Edition: Rendering issues with menu spell effects*: https://github.com/doitsujin/dxvk/issues/1943

NightHammer1000, 2021-03-17:
I can confirm that these Issues are not Isolated in Linux. Same Issues with Games that run on Windows using DXVK
In my case its the Sims 2 Issue on Windows with DXVK on a RX 6900 XT.

Driver Version 21.2.3 Optional.

Joshua-Ashton, 2021-03-17:
Far Cry 5 is also broken https://github.com/doitsujin/dxvk/issues/1978

NEOAethyr, 2021-05-12:
I checked on fc5, this seems to break it:
https://github.com/doitsujin/dxvk/commit/c98c5f5d1720df09d584b8353bf6739e06e8ffcc

I was going to do a simple revert line by line, but I don't know this rate it might break it.

Nvidia 450.119.03
Kubuntu 21.04
Kernel 5.11.18-m2

tested:
error	dxvk-d7db413cac0c76371846d309a39413d23c85af9b	HEAD is now at d7db413c [dxvk] Introduce prepareImage
bad	dxvk-2787ba84506b12f44aeb7bde53f46f04d1bea140
bad	dxvk-c98c5f5d1720df09d584b8353bf6739e06e8ffcc	.. HEAD is now at c98c5f5d [d3d11] Add internal DXGI usage flags to textures
works	dxvk-a60916f7ee5968077b9478e91d169482bd3e89a5
works	dxvk-58d6f018bb12eef96afa3e1597f5e42b673e2221
works	dxvk-8de9dc93781b61a902a2dc15cc4192514a821237	HEAD is now at 8de9dc93 [d3d11] Enable d3d11.invariantPosition by default

Ignore the error, it's a compile error.. (v1.8 above it compiles and so does 1.7.3 below it)

I tried this and it obviously didn't work, no idea...
https://github.com/GloriousEggroll/proton-ge-custom/blob/master/patches/dxvk/farcry5-hotfix.patch

Joshua-Ashton, 2021-05-13:
@NEOAethyr Dunia games work whether the sun is shining or not or if it's raining in France.
There is no problem with that commit and it works fine on an MSVC build.

Not related to this issue.

NEOAethyr, 2021-05-15:
[image]
[image]
[image]

What I was trying to say but not clearly enough, there's a wackiness with fc5 in general, and it's not just amd.

https://forums.developer.nvidia.com/t/missing-textures-in-far-cry-5-of-all-versions-of-linux-nvidia-460-xx-drivers/173192
https://www.protondb.com/app/552520

Tbh I didn't realize I posted in a AMDVLK thread until after the fact.
The other thread was closed.
And I assume if I made another it would of been seen as a dupe.

The video with the amd is a little different but there's wash out on both.
It seems kinda simular.
I sat there and cloned every rev to to find the last one it worked on (ran out of ram and crashed once thinking I would just build every rev..).
At least on my end, and apperently several others though they're blaming the nv driver.
I stayed away from the 460 drivers just to try to prevent that..

I don't know, my bad.

It's not like it's my lod mods, it's the stock release...
For a bit there I thought it was some obscure point update to the nv driver.
Between 1.7.3 and 1.8, the screens are from 1.8.1 I just took a min ago.
1.7.3 works and I noted the last commit where it does above..

I can't do no msvc build on my end, I just don't want to deal with that (windows and vc blah blah whatever ver it is these days)...
If the release is msvc compiled, well I don't know, 'cause it doesn't work as intended.. ^^
It's not that.
It's whatever that commit was, at least on my setup..

Joshua-Ashton, 2021-05-15:
Sure, but we are aware it's broken. The game's behaviour changes depending on whatever random stack values were in the right places in random functions. (ie. game is probably fucked)

jinjianrong, 2021-08-02:
Please try 2021.Q3.2 build

closed 2021-08-02 by jinjianrong

NEOAethyr, 2021-08-22:
Sorry for getting back late, it's the 2 factor thing that puts me off logging in.

The latest nvidia driver fixed the prob.
Also, I was incorrect before on the working dxvk ver anyways, if I would of just turned around in the game and walked a bit I would of seen more glitches.
But all good now.