GPUOpen-Drivers/AMDVLK #211: shader cache of amdvlk-pro 20.50 Windows driver is ignored after system reboot

aufkrawall, 2021-03-23:
With Adrenalin 21.3.1 Windows driver ( 20.50.03.01-210310a-365057E-RadeonSoftwareAdrenalin2020 ), the driver's Vulkan shader cache is ignored when the system is rebooted.

Example:
1. Start Strange Brigade Vulkan (seems any application is affected)
2. In `C:\Users\USER\AppData\Local\AMD\VkCache`, the file `56A893DC0A80BEC5A0080D3650F00A.parc` gets created.
3. Reboot Windows and start the game again.
4. The file `56A893DC0A80BEC5A0080D3650F00A.parc` is ignored
5. Instead the game compiles its shaders anew and the driver creates a file `56A893DC0A80BEC5A0080D3650F00A_1.parc`

This seems to be a regression of that driver version, it wasn't that bad in the past. Though it could also happen that under some circumstances (e.g. system configuration changes) the driver ignored the original cache file and generated a `samefilename_1.pac`. It would be nice if the driver really only ever disregarded the cached shaders when the driver or game binary actually have changed. It probably should never do that `_1.parc` thing, as the hashes of the application and driver/shader compiler apparently are still valid.

The 20.50 amdvlk-pro Linux driver doesn't show the regression mentioned above and also seems to be more resilient than the Windows driver regarding the `_1.parc` phenomenon in general. Tested on Windows 10 20H2 with an RX 6800.

jinjianrong, 2021-03-24:
@aufkrawall, Thanks for reporting the issue. We will take a look.  Moving forward, you could report Windows driver specific issues on https://community.amd.com/t5/opengl-vulkan/bd-p/opengl-and-vulkan-discussions.

aufkrawall, 2021-04-20:
Can do.

Yet it's unbelievable and very disappointing that it's still broken 21.4.1 driver...

aufkrawall, 2021-05-13:
I've also noticed that amdvlk Linux driver shader cache includes a hash based on the used Linux kernel in file names, which means that if you boot a different kernel, it once again ignores the already cached shaders entirely. Why does it do this? 

This is not true for RADV, it re-uses cached shaders when booting a different kernel. Seems to me like this is really bad practice by amdvlk, and I can imagine that such apparently unnecessary checks are prone to error and might even be responsible for the issue on Windows (which is btw. still not fixed with 21.5.1 driver, you've already released 3-5 drivers for Windows without working Vulkan shader cache).

aufkrawall, 2021-05-19:
Nice to see `Some Vulkan apps ignoring previously created shader cache and creating new duplicate cache` in the changelog of 2021.Q2.3. Hopefully you can manage to include it in the Windows driver in the not too distant future.

Though also with 2021.Q2.3, it still creates duplicates but, as was my experience with previous driver on Linux, not in a harmful way:

When I start Strange Brigade in Proton, it creates `4F9F51BCA5A879AF675B33C49ACA56DA.parc`. This is probably for the game's launcher (which is an additional .exe binary that probably uses D3D11 and thus DXVK). When I close the launcher and launch it again, there is still no duplicate created. But when I then run the game (native Vulkan) from the launcher, it creates `4F9F51BCA5A879AF72041B0C1AE0A380.parc` for the game (correct), but also `4F9F51BCA5A879AF675B33C49ACA56DA_1.parc` (duplicate of the cache for the launcher, but just a 84 bytes file stub).

qiaojbao, 2021-06-04:
I can reproduce this case.
When I change the vulkan driver, but I don't restart steam and re-create the proton command, I saw the 84byte _1.parc file.
And the game can not launch successfully.

After I restart steam, and re-create proton command, there are the right size _1.parc, and the game works.

The StrangeBrigade has 7 parc files while steam prepare the cache first time, because steam run 7 processes to do this.
ogl@ogl-desktop:/SteamGames/SteamLibrary/steamapps/shadercache/312670/AMDv1$ lsof *
COMMAND     PID USER   FD   TYPE DEVICE SIZE/OFF     NODE NAME
fossilize 12752  ogl   14uW  REG   8,27   558340 49810373 steamapp_shader_cache_2.parc
fossilize 12753  ogl   15uW  REG   8,27   523744 49810390 steamapp_shader_cache_6.parc
fossilize 12754  ogl   17uW  REG   8,27   712844 49810381 steamapp_shader_cache_3.parc
fossilize 12755  ogl   18uW  REG   8,27   606448 49807569 steamapp_shader_cache.parc
fossilize 12756  ogl   19uW  REG   8,27   544444 49810372 steamapp_shader_cache_1.parc
fossilize 12757  ogl   20uW  REG   8,27   587440 49810387 steamapp_shader_cache_5.parc
fossilize 12758  ogl   21uW  REG   8,27   586628 49810384 steamapp_shader_cache_4.parc

qiaojbao, 2021-06-08:
BTW, the game has a launcher dialog on the beginning, you can choose Vulkan or DX, then the game run.
For this reason, the launcher and game seem open the same cache file, and the game get a read-only access permission, so it creates a new one to update cache. 
The launcher:
COMMAND     PID USER   FD   TYPE DEVICE SIZE/OFF     NODE NAME
StrangeBr 13321  ogl   46uW  REG   8,27  3940472 49807569 steamapp_shader_cache.parc
StrangeBr 13321  ogl   47rW  REG   8,27  3943196 49810372 steamapp_shader_cache_1.parc
...
The game:
COMMAND     PID USER   FD   TYPE DEVICE SIZE/OFF     NODE NAME
StrangeBr 13532  ogl   46u   REG   8,27  3940472 49807569 steamapp_shader_cache.parc
StrangeBr 13532  ogl   47r   REG   8,27  3943196 49810372 steamapp_shader_cache_1.parc
...

aufkrawall, 2021-07-02:
Fixed with 21.6.2, just four months of breakage are finally over...

closed 2021-07-02 by aufkrawall