GPUOpen-Drivers/AMDVLK #214: Unpredictable results when calculating derivatives in helper invocations

ReinUsesLisp, 2021-04-02:
[This](https://github.com/GPUOpen-Drivers/AMDVLK/files/6251168/broken.zip) fragment shader generates unpredictable results on 21.2.2, Windows drivers.

The shader is calculating derivatives after a demote instruction is executed, which should give predictable results since helper invocations contribute to derivatives.

Replacing the `OpDemoteToHelperInvocationEXT` + `OpBranch` pair with `OpReturn` works around the issue, although it's incorrect and introduces more problems on other shaders.

Replacing `OpImageSampleImplicitLod` with `OpImageSampleExplicitLod` and a forced Lod of zero also works around the issue, but changes the behavior of the shader.

Recompiling the shader with `spirv-cross` -> `glslang` also fixes the issue.

I am seeing an `exp mrt0, off, off, off, off done vm` + `s_endpgm` pair before the derivatives are calculated and used by `image_sample`, although this also happens on the `glslang` output, that works as expected.

If more information or a better way to reproduce the issue is needed, please let me know.

jinjianrong, 2021-04-12:
@ReinUsesLisp could you provide the pipeline file?

ReinUsesLisp, 2021-04-13:
I've attached a dummy RGA pipeline, although it seems to store the absolute address to the shader source files. The pipeline has no special configuration.
[derivatives.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/6301429/derivatives.zip)

jinjianrong, 2024-05-22:
Per Vulkan spec:
If the image or sampler object used by an implicit derivative image instruction is not uniform across the quad and quadDivergentImplicitLod is not supported, then the derivative and LOD values are undefined.

closed 2024-05-22 by jinjianrong