GPUOpen-Drivers/AMDVLK #215: SyntaxError: invalid syntax when downloading source code.  file=sys.stderr

Randall-Coding, 2021-04-05:
When running this command  `repo init -u https://github.com/GPUOpen-Drivers/AMDVLK.git -b master
` as part of the installation process, I receive a syntax error in the python script main.py

> vulkandriver/.repo/repo/main.py", line 79
>     file=sys.stderr)
>         ^
> SyntaxError: invalid syntax

I have python3 installed.  I'm going to try removing the offending line and try again.  Any advice as to why this is happening?

Flakebi, 2021-04-05:
Maybe it’s running with python2 by accident? From the source code, `repo` wants at least python 3.6. Running with 3.5 prints a warning and everything below aborts.

baryluk, 2021-04-06:
What version of a repo tool are you using.

I don't have issues with repo 2.12-1 from Debian testing.

Randall-Coding, 2021-04-06:
Sorry for my late response.  Playing around with all these GPU settings really did a number on my machine (I think it was installing some of the proprietary Radeon software that tanked my machine).

I updated my kernel to 5.4 and updated Ubuntu from 18.04 to 20.04.  I will look into the points mentioned here and get back to you

Randall-Coding, 2021-04-07:
Now I'm following the Ubuntu 20.04 instructions.  I'm able to do most of the `repo sync` step now but it started to hang after 7/8.

> Fetching: 75% [3 jobs] (6/8) finished llpc
> Fetching: 87% [2 jobs] (7/8) finished pal
> ->hangs

Randall-Coding, 2021-04-07:
I tried deleting everything and starting over.  Looks like maybe the issue is just a slow download on the last repo.  Strangely the log messages don't show up on repo (even in verbose mode) until after I abort the operation.  Looks like a slow download speed could be the culprit.

Randall-Coding, 2021-04-07:
Left it running over night and it completed successfully.  Turns out it was just a long download.

> Fetching: 100% (8/8), done in 5h12m8.467s
> Updating files: 100% (101166/101166), done.
> Checking out: 100% (8/8), done in 15.842s
> repo sync has finished successfully.

baryluk, 2021-04-07:
> Fetching: 100% (8/8), done in 5h12m8.467s

Yes, the llvm one is very big.

Glad you have it working.

You were probably using too old repo or some weird manually installed repo version on the Ubuntu 18.04 before, that caused some issues.

Randall-Coding, 2021-04-07:
Unfortunately now I'm running into errors on one of the build commands that I'm trying to figure out.  

`ninja -C drivers/xgl/builds/Release64
`

Do you know if there is a way to re-build everything without having to do that long `repo sync` step again?
I think I may need a rebuild because my PC shutoff during the make process by accident.  

Here is the error output for reference.    

> Linking CXX shared library compiler/llpc/llvm/lib/libLTO.so.13git
> FAILED: compiler/llpc/llvm/lib/libLTO.so.13git 
> : && /usr/bin/c++ -fPIC -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wsuggest-override -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -O3 -DNDEBUG  -Wl,-z,defs -Wl,-z,nodelete  -s  -Wl,-rpath-link,/home/super/vulkandriver/drivers/xgl/builds/Release64/compiler/llpc/llvm/./lib  -Wl,-O3 -Wl,--gc-sections  -Wl,--version-script,"/home/super/vulkandriver/drivers/xgl/builds/Release64/compiler/llpc/llvm/tools/lto/LTO.exports" -shared -Wl,-soname,libLTO.so.13git -o compiler/llpc/llvm/lib/libLTO.so.13git compiler/llpc/llvm/tools/lto/CMakeFiles/LTO.dir/LTODisassembler.cpp.o compiler/llpc/llvm/tools/lto/CMakeFiles/LTO.dir/lto.cpp.o  -Wl,-rpath,"\$ORIGIN/../lib"  compiler/llpc/llvm/lib/libLLVMAMDGPUAsmParser.a  compiler/llpc/llvm/lib/libLLVMAMDGPUCodeGen.a  compiler/llpc/llvm/lib/libLLVMAMDGPUDesc.a  compiler/llpc/llvm/lib/libLLVMAMDGPUDisassembler.a  compiler/llpc/llvm/lib/libLLVMAMDGPUInfo.a  compiler/llpc/llvm/lib/libLLVMBitReader.a  compiler/llpc/llvm/lib/libLLVMCore.a  compiler/llpc/llvm/lib/libLLVMCodeGen.a  compiler/llpc/llvm/lib/libLLVMLTO.a  compiler/llpc/llvm/lib/libLLVMMC.a  compiler/llpc/llvm/lib/libLLVMMCDisassembler.a  compiler/llpc/llvm/lib/libLLVMSupport.a  compiler/llpc/llvm/lib/libLLVMTarget.a  compiler/llpc/llvm/lib/libLLVMAsmPrinter.a  compiler/llpc/llvm/lib/libLLVMDebugInfoDWARF.a  compiler/llpc/llvm/lib/libLLVMGlobalISel.a  compiler/llpc/llvm/lib/libLLVMSelectionDAG.a  compiler/llpc/llvm/lib/libLLVMMIRParser.a  compiler/llpc/llvm/lib/libLLVMAMDGPUDesc.a  compiler/llpc/llvm/lib/libLLVMAMDGPUInfo.a  compiler/llpc/llvm/lib/libLLVMAMDGPUUtils.a  compiler/llpc/llvm/lib/libLLVMCodeGen.a  compiler/llpc/llvm/lib/libLLVMPasses.a  compiler/llpc/llvm/lib/libLLVMTarget.a  compiler/llpc/llvm/lib/libLLVMCoroutines.a  compiler/llpc/llvm/lib/libLLVMipo.a  compiler/llpc/llvm/lib/libLLVMVectorize.a  compiler/llpc/llvm/lib/libLLVMFrontendOpenMP.a  compiler/llpc/llvm/lib/libLLVMIRReader.a  compiler/llpc/llvm/lib/libLLVMAsmParser.a  compiler/llpc/llvm/lib/libLLVMInstrumentation.a  compiler/llpc/llvm/lib/libLLVMScalarOpts.a  compiler/llpc/llvm/lib/libLLVMBitWriter.a  compiler/llpc/llvm/lib/libLLVMAggressiveInstCombine.a  compiler/llpc/llvm/lib/libLLVMExtensions.a  compiler/llpc/llvm/lib/libLLVMInstCombine.a  compiler/llpc/llvm/lib/libLLVMLinker.a  compiler/llpc/llvm/lib/libLLVMObjCARCOpts.a  compiler/llpc/llvm/lib/libLLVMTransformUtils.a  compiler/llpc/llvm/lib/libLLVMAnalysis.a  compiler/llpc/llvm/lib/libLLVMProfileData.a  compiler/llpc/llvm/lib/libLLVMObject.a  compiler/llpc/llvm/lib/libLLVMBitReader.a  compiler/llpc/llvm/lib/libLLVMCore.a  compiler/llpc/llvm/lib/libLLVMRemarks.a  compiler/llpc/llvm/lib/libLLVMBitstreamReader.a  compiler/llpc/llvm/lib/libLLVMMCParser.a  compiler/llpc/llvm/lib/libLLVMMC.a  compiler/llpc/llvm/lib/libLLVMDebugInfoCodeView.a  compiler/llpc/llvm/lib/libLLVMDebugInfoMSF.a  compiler/llpc/llvm/lib/libLLVMTextAPI.a  compiler/llpc/llvm/lib/libLLVMBinaryFormat.a  compiler/llpc/llvm/lib/libLLVMSupport.a  -lrt  -ldl  -lpthread  -lm  compiler/llpc/llvm/lib/libLLVMDemangle.a && :
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMAMDGPUCodeGen.a(AMDGPUSubtarget.cpp.o): in function `llvm::R600Subtarget::R600Subtarget(llvm::Triple const&, llvm::StringRef, llvm::StringRef, llvm::TargetMachine const&)':
> AMDGPUSubtarget.cpp:(.text._ZN4llvm13R600SubtargetC2ERKNS_6TripleENS_9StringRefES4_RKNS_13TargetMachineE+0x15e): undefined reference to `vtable for llvm::SelectionDAGTargetInfo'
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMAMDGPUCodeGen.a(AMDGPUSubtarget.cpp.o): in function `llvm::GCNSubtarget::GCNSubtarget(llvm::Triple const&, llvm::StringRef, llvm::StringRef, llvm::GCNTargetMachine const&)':
> AMDGPUSubtarget.cpp:(.text._ZN4llvm12GCNSubtargetC2ERKNS_6TripleENS_9StringRefES4_RKNS_16GCNTargetMachineE+0x19e): undefined reference to `vtable for llvm::SelectionDAGTargetInfo'
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMAMDGPUCodeGen.a(AMDGPUSubtarget.cpp.o): in function `llvm::R600Subtarget::~R600Subtarget()':
> AMDGPUSubtarget.cpp:(.text._ZN4llvm13R600SubtargetD2Ev[_ZN4llvm13R600SubtargetD5Ev]+0x2e): undefined reference to `llvm::SelectionDAGTargetInfo::~SelectionDAGTargetInfo()'
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMAMDGPUCodeGen.a(AMDGPUSubtarget.cpp.o): in function `non-virtual thunk to llvm::R600Subtarget::~R600Subtarget()':
> AMDGPUSubtarget.cpp:(.text._ZN4llvm13R600SubtargetD2Ev[_ZN4llvm13R600SubtargetD5Ev]+0x15e): undefined reference to `llvm::SelectionDAGTargetInfo::~SelectionDAGTargetInfo()'
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMAMDGPUCodeGen.a(AMDGPUSubtarget.cpp.o): in function `llvm::GCNSubtarget::~GCNSubtarget()':
[... cut ...]
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMSelectionDAG.a(ScheduleDAGVLIW.cpp.o):(.data.rel.ro._ZTVN12_GLOBAL__N_115ScheduleDAGVLIWE+0x40): undefined reference to `llvm::ScheduleDAGSDNodes::getGraphNodeLabel[abi:cxx11](llvm::SUnit const*) const'
> /usr/bin/ld: compiler/llpc/llvm/lib/libLLVMSelectionDAG.a(ScheduleDAGVLIW.cpp.o):(.data.rel.ro._ZTVN12_GLOBAL__N_115ScheduleDAGVLIWE+0x78): undefined reference to `llvm::ScheduleDAGSDNodes::getCustomGraphFeatures(llvm::GraphWriter<llvm::ScheduleDAG*>&) const'
> collect2: error: ld returned 1 exit status
> [45/635] Building CXX object compiler/llpc/CMakeFiles/llpc.dir/translator/lib/SPIRV/SPIRVReader.cpp.o
>

baryluk, 2021-04-07:
> Unfortunately I'm running into errors on one of the build commands that I'm trying to figure out.
> 
> `ninja -C drivers/xgl/builds/Release64 `

Just do `rm -rf drivers/xgl/builds/Release64` and redo the cmake + ninja steps. Next time open separate issue ticket for separate issues.

Randall-Coding, 2021-04-07:
Ok thanks again. Will do.

closed 2021-04-08 by Randall-Coding