GPUOpen-Drivers/AMDVLK #217: Unsupported image format on Star Swarm

libcg, 2021-04-07:
I'm currently working on [GRVK](https://github.com/libcg/grvk) which implements the Mantle API over Vulkan. The Star Swarm demo partially runs, however the game is trying to create images with the following properties:
- format: `VK_FORMAT_R32_UINT`
- type: `VK_IMAGE_TYPE_3D`
- tiling: `VK_IMAGE_TILING_LINEAR`
- usage: `VK_IMAGE_USAGE_TRANSFER_SRC_BIT | VK_IMAGE_USAGE_TRANSFER_DST_BIT`

AMDVLK doesn't support that combination, however 1D and 2D seem to work. I'd like to know if this could be fixed.

libcg, 2021-04-08:
I missed some, here's the full list:
- `R32_UINT` 3D linear `TRANSFER_SRC | TRANSFER_DST | SAMPLED_BIT | STORAGE_BIT`
- `BC4_UNORM_BLOCK` 2D linear `TRANSFER_SRC | TRANSFER_DST`
- `BC7_UNORM_BLOCK` 2D linear `TRANSFER_SRC | TRANSFER_DST`

qiaojbao, 2021-06-10:
I write a simple code to create an image with your create-info in vkcube, there is no error.
Could you give more detail about your demo code?

closed 2024-05-22 by jinjianrong