GPUOpen-Drivers/AMDVLK #219: /usr/lib/i386-linux-gnu/amdvlk32.so: wrong ELF class

Randall-Coding, 2021-04-08:
Ubuntu 20.04, Kernel 5.8.0, 64-bit

Ran vulkaninfo and it said the following

> ERROR: [Loader Message] Code 0 : /usr/lib/i386-linux-gnu/amdvlk32.so: cannot open shared object file: 
> No such file or directory
> WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
> ERROR at /build/vulkan-tools-1.2.170.0~rc2-1lunarg20.04/vulkaninfo/vulkaninfo.h:248:vkEnumeratePhysicalDevices failed with ERROR_INITIALIZATION_FAILED

I had thought since my system is 64bit I wouldn't need to use 32?  

Anyway I go back to the instructions and see they changed.  I try the command to build Release32 for Ubuntu 20.04

I receive an error on `include openssl/opensslconf.h`

> In file included from /usr/include/openssl/sha.h:13,
>                  from /home/super/vulkandriver/drivers/pal/src/util/lnx/lnxHashProvider.h:36,
>                  from /home/super/vulkandriver/drivers/pal/src/util/lnx/lnxHashProvider.cpp:32:
> /usr/include/openssl/e_os2.h:13:11: fatal error: openssl/opensslconf.h: No such file or directory
>    13 | # include <openssl/opensslconf.h>
>       |           ^~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> [2175/2326] Building CXX object pal/src/CMakeFiles/pal.dir/util/lnx/lnxArchiveFile.cpp.o
> 

So I made a blank opensslconf.h file and re-ran the build command which appeared to work. 

Then I copied over the amdvlk32.so file into the (newly created) /usr/lib/i386-linux-gnu/ folder and ran vulkaninfo again.  New error.

> ERROR: [Loader Message] Code 0 : /usr/lib/i386-linux-gnu/amdvlk32.so: wrong ELF class: ELFCLASS32
> WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
> ERROR at /build/vulkan-tools-1.2.170.0~rc2-1lunarg20.04/vulkaninfo/vulkaninfo.h:248:vkEnumeratePhysicalDevices failed with ERROR_INITIALIZATION_FAILED

I've re-read the instructions a few times, but not sure where I went wrong.  I did skip the part about "to make AMDVLK driver work correctly on system with both AMDVLK and RADV installed" but is that optional?  

Also should I have already had an /etc/vulkan folder from a previous install?  I didn't have one.

Randall-Coding, 2021-04-08:
Maybe I need to ask some fundamental questions.  This driver replaces the Radeon proprietary software correct?  I'm trying to use the open driver because the Radeon proprietary software is no longer supported for HD 7730 and trying to install that was breaking my machine anyway.  Just want to make sure this is even possible to achieve for me.

jinjianrong, 2021-04-08:
RADV is installed by default on Ubuntu 20.04, so you need to disable it or enable the AMD switchable graphics layer. 
HD 7730 is supposed to be supported by amdvlk but it is not well tested.
BTW, you need to enable amdgpu.si_support and amdgpu.cik_support  in kernel

Randall-Coding, 2021-04-08:
Thanks.  

Last night I tried looking into disabling RADV, but couldn't find any instructions on that.  I found a few pages talking about changing amdgpu.si_support, which I di, and that creates a weird loop on the login screen (I try to login, and it brings me right back to the login screen).  Here is how I changed my grub startup

> GRUB_CMDLINE_LINUX_DEFAULT="radeon.si_support=0 radeon.cik_support=0 amdgpu.si_support=1 amdgpu.cik_support=1 amdgpu.dc=1 amdgpu.dpm=1 amdgpu.modeset=1"  

Although now I've removed some components of the original amdvlk setup, so I can try adding those back with the new grub setup.

closed 2024-05-22 by jinjianrong