GPUOpen-Drivers/AMDVLK #222: AMD_VULKAN_ICD has no effect when both RADV and AMDVLK are installed

xuzhao9, 2021-04-16:
After installing amdvlk and radeon (libvulkan_radeon) driver together, the vulkan still loads RADV driver by default.
I have placed the amd_icd.x86_64.json in /etc/vulkan/implicit_layer.d, and set the environment variable AMD_VULKAN_ICD to "AMDVLK", however, the vkcube program still uses RADV.

I don't find anything related to "AMD_VULKAN_ICD" environment variable in [vulkan loader](https://github.com/KhronosGroup/Vulkan-Loader), so I am curious where is this environment variable respected and checked?

Platform: openSUSE Tumbleweed 20210307

jinjianrong, 2021-04-19:
AMD_VULKAN_ICD is checked in https://github.com/GPUOpen-Drivers/xgl/blob/dev/icd/layers/vk_layer_switchable_graphics.cpp.
Is the amd_icd.x86_64.json file generated according to the instruction:  https://github.com/GPUOpen-Drivers/AMDVLK#build-driver-and-generate-json-files?

xuzhao9, 2021-04-19:
Yes, it is generated by the ninja compilation program. To be specific, this is the amd_icd.x86_64.json file I am using: https://pastebin.com/NZ9ZVSR4

Flakebi, 2021-04-19:
Does the switchable graphics layer and amdvlk show up in the output of `vulkaninfo`?

xuzhao9, 2021-04-19:
> Does the switchable graphics layer and amdvlk show up in the output of `vulkaninfo`?

Here is the output of `vulkaninfo`: https://pastebin.com/Lu1DgUcb. AMD switchable graphics layer appears but I don't find amdvlk.

Flakebi, 2021-04-19:
Thanks for the output. That looks like it is already using amdvlk (`GPU id : 0 (Unknown AMD GPU)`).
You can try `AMD_VULKAN_ICD=RADV vulkaninfo` and it should show a GPU that has RADV (and ACO) in its name.

xuzhao9, 2021-04-19:
> Thanks for the output. That looks like it is already using amdvlk (`GPU id : 0 (Unknown AMD GPU)`).
> You can try `AMD_VULKAN_ICD=RADV vulkaninfo` and it should show a GPU that has RADV (and ACO) in its name.

Indeed as you said, after specifying `AMD_VULKAN_ICD=RADV`, vulkaninfo shows a GPU with RADV in its name.
However, when I run `AMD_VULKAN_ICD=AMDVLK vkcube`, the program gives the following output:
```
WARNING: radv is not a conformant vulkan implementation, testing use only.
Selected GPU 0: Unknown AMD GPU, type: 2
```
Is it expected behavior?

Flakebi, 2021-04-19:
Ah, this can be a bit confusing, indeed.
The line `Selected GPU 0: Unknown AMD GPU, type: 2` says it is using amdvlk as the vulkan driver.
The line `WARNING: radv is not a conformant vulkan implementation, testing use only.` (which I’d say is a bit of a stretch, given the usage of radv ;)) is printed when the driver libraries get loaded, that is before the application chooses which driver to use and it doesn’t mean that radv is actually used, just the .so was loaded by the vulkan loader.

xuzhao9, 2021-04-19:
> Ah, this can be a bit confusing, indeed.
> The line `Selected GPU 0: Unknown AMD GPU, type: 2` says it is using amdvlk as the vulkan driver.
> The line `WARNING: radv is not a conformant vulkan [...]

I see, now it makes sense. Thanks!

closed 2021-04-19 by xuzhao9