GPUOpen-Drivers/AMDVLK #223: amdPalSettings.cfg missing

ripper81, 2021-04-17:
I have installed AMDVLK on Kubuntu 20.10. I'd like to use AMD Freesync with it. I know that i have to edit the amdPalSettings.cfg file to enable it. The problem is that i can't find the config file on my system.

Flakebi, 2021-04-17:
You have to create it yourself in `~/.config`. Also, I think the file is now called `amdVulkanSettings.cfg`, although `amdPalSettings.cfg` works too.
I think freesync should work out of the box, at least it does for me.

ripper81, 2021-04-17:
I don't know how to verify if Freesync is working. All i know is that my games are stuttering with AMDVLK while they run flawless with MESA RADV with Freesync enabled. Overall the performance is far lower. I also hate the fact that my system now defaults to AMDVLK. It should be the other way around because i only need AMDVLK for one game (Metro Exodus) that crashes with MESA RADV. I don't know why we even need two opensource vulkandrivers for one OS, where one is supported by AMD but by no game or application and the other is widely supported by software but not by the company selling the hardware.

closed 2024-05-22 by jinjianrong