GPUOpen-Drivers/AMDVLK #228: AMDVLK crashes kernel driver in Strange Brigade if the game has been started with RADV before

aufkrawall, 2021-05-16:
-Start Strange Brigade (Vulkan) with RADV, then close it and start it again with AMDVLK: amdgpu kernel driver crashes and resets upon reaching the game's main menu.
-Start it first with AMDVLK and then with RADV: amdgpu kernel driver doesn't crash.
-Start it two times in a row with AMDVLK: It doesn't crash.
-Start it two times in a row with RADV: It doesn't crash.

So, it seems RADV puts something in a state that AMDVLK doesn't like. Ping @pendingchaos . :)

Errors in dmesg:
```
[ 2278.493879] [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
[ 2283.622891] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx_0.0.0 timeout, signaled seq=227004, emitted seq=227006
[ 2283.622980] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process StrangeBrigade_ pid 151480 thread StrangeBrigade_ pid 151480
[ 2283.623056] amdgpu 0000:03:00.0: amdgpu: GPU reset begin!
[ 2283.962222] amdgpu 0000:03:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
[ 2283.962288] [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KGQ disable failed
[ 2284.184339] amdgpu 0000:03:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
[ 2284.184400] [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KCQ disable failed
[ 2284.406005] [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* failed to halt cp gfx
[ 2284.425024] [drm] free PSP TMR buffer
[ 2284.473807] amdgpu 0000:03:00.0: amdgpu: MODE1 reset
[ 2284.473875] amdgpu 0000:03:00.0: amdgpu: GPU smu mode1 reset
[ 2284.981885] amdgpu 0000:03:00.0: amdgpu: GPU reset succeeded, trying to resume
```

[dmseg.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/6489071/dmseg.log)

Happens with both amdvlk-open 2021.Q2.2 and amdvlk-pro 21.10. Linux 5.12.4, Proton Experimental.

Oschowa, 2021-05-18:
I tried this on Polaris and couldn't reproduce it there.

iWeaker4you, 2021-05-18:
I am trying to reproduce it and I have not been able to do it.

aufkrawall, 2021-05-18:
With an RDNA2 GPU?

pingubot, 2021-05-22:
I had very strange gpu hangs in many games with amdvlk and amd-pro (Strange Brigade and dx12 games via vkd3d).They are all gone now when i do not use radv before. I have a 6900xt.

aufkrawall, 2021-05-22:
Yeah, I think it can happen with all sort of applications. Hopefully this reproducible case covers them all.

pingubot, 2021-05-23:
> Yeah, I think it can happen with all sort of applications. Hopefully this reproducible case covers them all.

Yep. And many thx for figuring out the cause of those hangs !

pingubot, 2021-06-25:
That still seems to be an issue with the recent release q26.

NTMan, 2021-07-14:
I am able reproduce it bug with 6900XT and several games such as Zombie Army 4, Resident Evil Village, Forza Horizon 4, Avengers.
https://gitlab.freedesktop.org/mesa/mesa/-/issues/5055

closed 2024-05-22 by jinjianrong