GPUOpen-Drivers/AMDVLK #229: Above 4G Decoding comes with a strong CPU bound performance penalty in Red Dead Redemption 2

aufkrawall, 2021-05-19:
Run the benchmark and wait for the desert scene. Without Above 4G Decoding, the game is GPU bound and runs at ~100fps all the time:
[image]

When Above 4G Decoding is enabled in bios, the same scene now is CPU bound at ~65fps for most of the time:
[image]

It also happens on Windows when resizable BAR is enabled in bios. I could also reproduce on another system with X470 Taichi mainboard + Ryzen 5600X.

RX 6800
I5 11400F + ASRock B560 Pro4 (bios 1.80)
Linux 5.12.5
amdvlk-open 2021.Q2.3 and amdvlk-pro 21.10
wine-staging 6.8
latest Red Dead Redemption 2 version in Rockstar Games Launcher (native Vulkan)
set game to 720p render resolution and e.g. "balanced" graphics preset

RafaelAmauri, 2022-01-17:
You're a godsent. I was getting terrible performance in this game and didn't know why. Disabling S.A.M. on my BIOS settings gave me 20-30 extra FPS for some reason. Using an RX 6600

mygitanon, 2022-02-17:
Can confirm that this is still an issue. With a 3080 (not updated with REBAR firmware) and 5600X, I was dropping to the 40s in Valentine. Changing res down to 1080P made no difference, was 100% CPU bound. Disabled above 4G decoding and voila, sitting pretty in the 80s-90s at 4K (DLSS Quality) in Valentine.

TJ-Hunt, 2022-07-15:
I know this is an old post but Resizable Bar is ONLY FOR NVIDIA. You will for sure have problems having it enabled on AMD. Resizable Bar is an NVIDIA tech.

TJ-Hunt, 2022-07-15:
> > I know this is an old post but Resizable Bar is ONLY FOR NVIDIA. You will for sure have problems having it enabled on AMD. Resizable Bar is an NVIDIA tech.
> 
> This is really not accurate. >256MB host visible VRAM is part of newer PCIe standards and enables applications and drivers to perform some optimizations with this limitation gone. There are also reports of issues with SAM with pre X570/B550 chipset platforms, also with Radeons. Though, it's hard to identify a more accurate pattern.

It is 100% accurate. While it "MAY" work its not designed for AMD. Period. "shrug" 4g Decoding is the AMD juice right now.

Not to say you're wrong so let me Clarify. When i say AMD and NVIDIA I am strictly talking about Video cards. Not motherboard chipsets.

RafaelAmauri, 2022-07-15:
> > > I know this is an old post but Resizable Bar is ONLY FOR NVIDIA. You will for sure have problems having it enabled on AMD. Resizable Bar is an NVIDIA tech.
> > 
> > This is really not accurate. >256MB host visible VRAM is part of newer PCIe standards and enables applications and drivers to perform some optimizations with this limitation gone. There are also reports of issues with SAM with pre X570/B550 chipset platforms, also with Radeons. Though, it's hard to identify a more accurate pattern.
> 
> It is 100% accurate. While it "MAY" work its not designed for AMD. Period. "shrug" 4g Decoding is the AMD juice right now.
> 
> Not to say you're wrong so let me Clarify. When i say AMD and NVIDIA I am strictly talking about Video cards. Not motherboard chipsets. Some motherboards may have issues because of poor firmware. That being said way more work as intended.

What are you contributing to this thread exactly? You're not here to submit a bug report, you're not here to suggest a solution and even less to provide one.

You're just here to be pedantic about the nomenclature of the implementation when both are exactly the same. Please read about enabling Smart Access Memory on the motherboard's manufacturer website before commenting further, thank you.

TJ-Hunt, 2022-07-15:
More than you. Period. The main post is a guy with an amd card and amd board complaining about a Nvidia utility messing his fps up. Smfh stay in school bruh

Faugus, 2022-12-27:
Any news for this? It happens with me too, but on Uncharted.

RafaelAmauri, 2022-12-27:
What solved it for me was switching my motherboard.

It seems to be an issue with some models of the A-520. Sold it and bought a
B450 and the problem is gone. I even bought uncharted and am running it
without any issues

Faugus, 2022-12-27:
I tried 2 different mobos (both Asus) and I'm having this problem

Bunkerbewohner, 2023-02-12:
For the record: I just stumbled upon this issue when I was looking for reasons why Hogwarts Legacy was running like dog shit (<60 FPS even on low settings, depending on the area) on my AMD system (Ryzen 9 3900X + Radeon RX 3950 XT). Turns out I had "Above 4G Encoding" enabled in my BIOS. Disabled that and the FPS increased _dramatically_ and now runs smoothly with 100+ FPS (High settings, 1440p).

Faugus, 2023-02-12:
> For the record: I just stumbled upon this issue when I was looking for reasons why Hogwarts Legacy was running like dog shit (<60 FPS even on low settings, depending on the area) on my AMD system (Ryzen 9 3900X + Radeon RX 3950 XT). Turns out I had "Above 4G decoding" enabled in my BIOS. Disabled that and the FPS increased _dramatically_ and now runs smoothly with 100+ FPS (High settings, 1440p).

Yeah... in some games, Resizable BAR just doesn't work correctly in Linux.
I tried everything and I couldn't get it to work.
It's not a hardware problem, cause I tested on different systems (B450 and B550).
Red Dead Redemption runs perfectly fine for me.
I think it's a VKD3D problem, cause I didn't see this problem happening on DXVK and Vulkan Native games.

Radeuz, 2023-02-22:
RX 5600 XT, Ryzen 5 3600 on B450 motherboard. Having the same issue on rdr2. %100 CPU Usage and low GPU usage when i enable Smart Access Memory.

Faugus, 2023-02-25:
For VKD3D games, running with `VKD3D_CONFIG=no_upload_hvv` may solve the issue.
It fixed the problem I was having with Uncharted.

Here's a comparison:
https://www.youtube.com/watch?v=78XakTs-3bI

`VKD3D_CONFIG=no_upload_hvv` - Blocks any attempt to use host-visible VRAM (large/resizable BAR) for the UPLOAD heap. May free up vital VRAM in certain critical situations, at cost of lower GPU performance. A fraction of VRAM is reserved for resizable BAR allocations either way, so it should not be a real issue even on lower VRAM cards.
https://github.com/HansKristian-Work/vkd3d-proton

Faugus, 2023-02-25:
Here's a comparison running the game with ReBAR OFF, ReBAR ON and ReBAR ON + `VKD3D_CONFIG=no_upload_hvv`
https://www.youtube.com/watch?v=78XakTs-3bI

`VKD3D_CONFIG=no_upload_hvv` - Blocks any attempt to use host-visible VRAM (large/resizable BAR) for the UPLOAD heap. May free up vital VRAM in certain critical situations, at cost of lower GPU performance. A fraction of VRAM is reserved for resizable BAR allocations either way, so it should not be a real issue even on lower VRAM cards.
https://github.com/HansKristian-Work/vkd3d-proton

jinjianrong, 2024-03-25:
Should already be fixed with new release

closed 2024-03-25 by jinjianrong