GPUOpen-Drivers/AMDVLK #230: AMDPRO-21.10 and AMDVLK 2021.q23 Memory Leak with Borderlands 3 DX12

pingubot, 2021-05-22:
Hi,

when using amd-pro driver 21.10 i get a system hang during the "optimizing shaders" game screen. The reason is that the ram usage is growing and growing. I had to kill it while it was using nearly 30G of RAM to avoid another system hang, and the shader optimization was not not finished yet.

I tried the same with amdvlk 2021.q23 and it also leaks memory here, even the startup for the game takes very long and it already used 20G of RAM afterwards.

Use Borderlands 3 via Proton-experimental in steam.
Use DX12 in game, max settings.

```
inxi -GSC -xx
System:    Host: gamebox Kernel: 5.12.5-tkg x86_64 bits: 64 compiler: gcc v: 10.3.0 Desktop: KDE Plasma 5.21.5 
           tk: Qt 5.15.2 wm: kwin_x11 dm: SDDM Distro: openSUSE Tumbleweed 20210519 
CPU:       Info: 8-Core model: AMD Ryzen 7 3700X bits: 64 type: MT MCP arch: Zen 2 rev: 0 cache: L2: 4 MiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 115090 
           Speed: 4075 MHz min/max: 2200/3600 MHz boost: enabled Core speeds (MHz): 1: 4075 2: 4066 3: 4068 4: 4068 
           5: 4046 6: 4067 7: 4064 8: 4064 9: 4065 10: 4064 11: 4063 12: 4069 13: 4064 14: 4082 15: 4069 16: 4069 
Graphics:  Device-1: Advanced Micro Devices [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] driver: amdgpu v: kernel 
           bus-ID: 0a:00.0 chip-ID: 1002:73bf 
           Display: x11 server: X.Org 1.20.11 compositor: kwin_x11 driver: loaded: amdgpu resolution: 3440x1440 s-dpi: 96 
           OpenGL: renderer: AMD Radeon RX 6900 XT (SIENNA_CICHLID DRM 3.40.0 5.12.5-tkg LLVM 12.0.0) v: 4.6 Mesa 21.1.0 
           direct render: Yes
```

Many thx
Christian

jinjianrong, 2021-05-24:
@pingubot I checked the source code, EXT_calibrated_timestamps is exposed in amdvlk. could you double check if the extension is really not supported on your system?

DadSchoorse, 2021-05-24:
Wine uses `CLOCK_MONOTONIC_RAW` to implement Windows Performance Counters, so we need `VK_TIME_DOMAIN_CLOCK_MONOTONIC_RAW_EXT` to implement `VK_TIME_DOMAIN_QUERY_PERFORMANCE_COUNTER_EXT` in winevulkan.

 AMDVLK only supports the `DEVICE_EXT` and `CLOCK_MONOTONIC_EXT` domains at the moment, so everything that needs performance counter timestamps won't work.

closed 2024-05-22 by jinjianrong