GPUOpen-Drivers/AMDVLK #231: amd-pro 21.10 and amdvlk 201.q23 very slow in Ghostrunner demo DX12 and Control DX12

pingubot, 2021-05-22:
Hi,

for Control and Ghostrunner Demo the performance with DX12 is very bad.

Control:
AMd-Pro and amd-vlk: ~ 60-65fps
Radv: ~104 fps

Ghostrunner Demo:
AMd-Pro and amd-vlk: ~ 60-70fps
Radv: ~130 fps

Both titles with proton-experimental, 3440x1440, max settings. For The Ghostrunner demo you need a vkd3d version without calibrated timestamps requirement, looks like amd-pro and amdvlk do not support that feature. Ping me if you need a precompiled version.

```System:    Host: gamebox Kernel: 5.12.5-tkg x86_64 bits: 64 compiler: gcc v: 10.3.0 Desktop: KDE Plasma 5.21.5 
           tk: Qt 5.15.2 wm: kwin_x11 dm: SDDM Distro: openSUSE Tumbleweed 20210519 
CPU:       Info: 8-Core model: AMD Ryzen 7 3700X bits: 64 type: MT MCP arch: Zen 2 rev: 0 cache: L2: 4 MiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 115090 
           Speed: 4075 MHz min/max: 2200/3600 MHz boost: enabled Core speeds (MHz): 1: 4075 2: 4066 3: 4068 4: 4068 
           5: 4046 6: 4067 7: 4064 8: 4064 9: 4065 10: 4064 11: 4063 12: 4069 13: 4064 14: 4082 15: 4069 16: 4069 
Graphics:  Device-1: Advanced Micro Devices [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] driver: amdgpu v: kernel 
           bus-ID: 0a:00.0 chip-ID: 1002:73bf 
           Display: x11 server: X.Org 1.20.11 compositor: kwin_x11 driver: loaded: amdgpu resolution: 3440x1440 s-dpi: 96 
           OpenGL: renderer: AMD Radeon RX 6900 XT (SIENNA_CICHLID DRM 3.40.0 5.12.5-tkg LLVM 12.0.0) v: 4.6 Mesa 21.1.0 
           direct render: Yes
```

Many thx,
Christian

aufkrawall, 2021-05-23:
I can confirm this also with Hitman 2, Horizon Zero Dawn and Strange Brigade (D3D12). Thus I suspect this to be a very general issue AMDVLK has with VKD3D-Proton.

jinjianrong, 2021-06-11:
As the first step to investigate the issue, we are adding  CLOCK_MONOTONIC_RAW support for calibrated timestamps

pingubot, 2021-06-11:
Ok, nice for bringing that in. Control should work without that feature nonetheless, so if you want to take a look you can already start with that game.

pingubot, 2021-06-25:
Can confirm that Ghostrunner now starts with the recent amdvlk release due to  CLOCK_MONOTONIC_RAW now working , but as expected the performance is still bad.

pingubot, 2021-09-16:
any news on that topic ?

jinjianrong, 2021-09-23:
The performance issue is because VK_VALVE_mutable_descriptor_type is not supported by AMDVLK. We plan to implement the EXT version extension

pingubot, 2021-11-13:
Ayn news on this one ?

pingubot, 2021-12-10:
Another release, still with the extension missing. Any news when it could show up ?

Degerz, 2021-12-10:
Progress is likely blocked behind Khronos Group and no members are authorized to share detailed information on functionality prior to public release ...

pingubot, 2022-06-05:
Any news on that topic ?

bluestang2006, 2022-09-14:
The new v2022.Q3.4 have the same issue (slow) as well as the amdpro drivers.

pingubot, 2022-09-17:
> The performance issue is because VK_VALVE_mutable_descriptor_type is not supported by AMDVLK. We plan to implement the EXT version extension

Hi,

as the ext is available now, can we expect it to be implemented with the next release ?

Many thx !

bluestang2006, 2022-09-30:
> The performance issue is because VK_VALVE_mutable_descriptor_type is not supported by AMDVLK. We plan to implement the EXT version extension

2 releases and still not implemented?

jinjianrong, 2022-10-11:
The extension is under implementation, some code will show up in following releases, but we still need some effort to fix CTS failure to fully enable it.

bluestang2006, 2023-02-17:
> The extension is under implementation, some code will show up in following releases, but we still need some effort to fix CTS failure to fully enable it.

any updates on enabling `VK_EXT_mutable_descriptor_type`?

Degerz, 2023-02-25:
The feature hasn't been merged yet in the [compiler](https://github.com/GPUOpen-Drivers/llpc/pull/2243) ...

bluestang2006, 2023-03-17:
> The feature hasn't been merged yet in the [compiler](https://github.com/GPUOpen-Drivers/llpc/pull/2243) ...

its been merged in llpc, but I see that the extension isn't supported/exposed yet in xgl. Any progress with CTS fails?

bluestang2006, 2023-04-19:
**Update:** 
From the Q2023.2.1 branch point, I exposed the extension and ran the VK-CTS tests and they seem to pass. I also tested Control w/ RT and it is finally playable. However, there are issues like the occasional GPU hang so I imagine AMD is busy triaging those issues and tweaking perf. (its slower than RADV in Control)

Degerz, 2023-05-24:
With the release of mutable descriptors extension functionality, the author of this issue can test to see if the issue is solved and if so may close it ...

jinjianrong, 2023-05-25:
Thanks all for the update. Please help create new issue if you still see any quality or performance problem with latest release

closed 2023-05-25 by jinjianrong