GPUOpen-Drivers/AMDVLK #233: Visual corruption in DOOM Eternal with re-size bar enabled

DoomPenguin9, 2021-06-01:
As the title says, there are visual issues in DOOM Eternal when re-size bar is enabled. They occur with both AMDVLK and pro but they happe much quicker and more often with AMDVLK-open.  With the pro driver I can sometimes play for hours without this issue appearing but with the open driver it appears after 20-30 minutes at most. This issue does not occur when with RADV.

[image]
[image]
[image]
[image]

OdinVex, 2022-08-19:
I have the same issue, except mine also extends to many splash graphics, such as when starting any Challenge and the ripping of a heart happens.

DoomPenguin9, 2024-04-01:
This no longer happens for me with 2024.Q1.3-1

closed 2024-04-01 by DoomPenguin9