GPUOpen-Drivers/AMDVLK #238: How to install Driver to customized directroy rather than root？

xiaoqiang-wang, 2021-06-28:
Dear, my friends,
### I DO NOT want to install the Driver as root, I just want to install it under myself work directory. 
Just like python2 and python3 can be installed to general user customized directory.

#

I'm trying to compile and install AMD Open GPU Driver on my Ubuntu.
After I initialize the system environment and  sync code to my work_dir.  

I run build and install command as below:
### 
    Step 0. /home/andy/work_path$  cmake -G Ninja -S xgl -B xgl/builds/Release64  

after Step 0,  I think I DO NOT want install it as root.  
I just want install it under my work_dir rather than root. 
So I customize install directory with cmake argument  "-DCMAKE_INSTALL_PREFIX="./install".
### 
    Step 1. /home/andy/work_path$  cmake -G Ninja -S xgl -B xgl/builds/Release64 -DCMAKE_BUILD_TYPE=Debug -DLLVM_PARALLEL_LINK_JOBS=8 -DCMAKE_INSTALL_PREFIX="./install"

###
    Step 2 ./home/andy/work_path$ cmake --build xgl/builds/Release64

After Step 2, I see the "amdvk.so" and "amdllpc" binary files have successfully generated. then I try to run Step 3.

###
    step 3 ./home/andy/work_path$ cmake --install xgl/builds/Release64 --component icd

# Errors
 when I try Step 3 to install it, It report below error,  It still need root privileges.
###
    CMake Error at xgl/builds/Release64/cmake_install.cmake:54 (file):   file cannot create directory: /etc/vulkan/icd.d.  Maybe need administrative  privileges.

# Questions
### Q1: Can I install the driver as  "general user", rather than root?  How can I customize the install directory.

#
Before run do Step 1, I have  run the default command in Step 0.
### Q2: Does this problem caused by I not clean the default setting? How I can clean all default setting and customize my install directory.

Sincerely
xiaoqiang-wang

JaxLinAMD, 2021-06-28:
Hi @xiaoqiang-wang, the way you use "-DCMAKE_INSTALL_PREFIX="./install" is correct, but some files must be installed to /etc/vulkan/icd.d even though you choose to install to customized directory.
 
Could you try if "sudo cmake --install xgl/builds/Release64 --component icd" works?

xiaoqiang-wang, 2021-06-28:
Dear Lin,
🥇    I thanks your kindly reply very much.
This machine is used by several users,  so I DO NOT want my install will change system config.

😀 I have got what I wanted "amdvk.so" and "amdllpc" binary files after Step 2.
and currently on learning  "amdllpc" workflow.
So, I stop after Step 2. 

when one day, I got a machine for myself. 
I will try  the install command "sudo cmake --install xgl/builds/Release64 --component icd".

Sincerely
xiaoqiang-wang

Flakebi, 2021-06-28:
Hi,
If you want to use your compiled driver, you can edit the `amd_icd64.json` file (which should be in the same directory as `amdvlk64.so`) and change the path in `library_path` to your compiled `amdvlk64.so` (relative paths like `./amdvlk64.so` also work).
When running Vulkan applications, start them with `VK_ICD_FILENAMES=/path/to/your/amd_icd64.json` (e.g. `VK_ICD_FILENAMES=/home/andy/work_path/xgl/builds/Release64/icd/amd_icd64.json vkcube` or set this variable in your `~/.profile` or `~/.bashrc`) to use this driver.

xiaoqiang-wang, 2021-06-29:
Dear @Flakebi ，
      thanks a lot my friend.
 👍  you are really expert on the Driver useage.  
💯 I think your solution is what I wanted.

#
But I still have a qusetion,
#### Can I get an Open-Source AMDGPU simulator that can co-work with above Driver?

Sincerely
xiaoqiang-wang

JaxLinAMD, 2021-06-29:
@Flakebi  thanks for the help, learned new knowledge :)

closed 2021-06-29 by JaxLinAMD

Flakebi, 2021-06-29:
> Can I get an Open-Source AMDGPU simulator that can co-work with above Driver?

I’m not aware of any simulator. The closest thing is probably the “Infinitely Fast Hardware” mode: https://github.com/GPUOpen-Drivers/AMDVLK#runtime-settings (I never used that so far, so I don’t know what exactly it does)

dstutt, 2021-06-29:
IFH mode drops all calls before they are actually sent to the GPU hardware. This is most useful when evaluating driver performance (pretending that the hardware is infinitely fast, hence the name), or determining if there's a bug (crash) in driver side code (rather than anything running on the GPU itself).

xiaoqiang-wang, 2021-06-30:
Hello, @dstutt my friend,
  😀  Thanks for your kindly reply.
can you give me some suggestion, how to get a Simulator (not on GPU hardware, not drops draw-calls) that can co-work with above Driver?

#### Does the corresponding Simulator is part of intellectual property of AMD,  not published on github?

jinjianrong, 2021-06-30:
Simulator  is not published

xiaoqiang-wang, 2021-06-30:
😃  Hello @jinjianrong  my friend, 
thanks for your reply very much.

Will it be published in future?
Can I find a old version Simulator?

jinjianrong, 2021-07-06:
I don't think simulator will be published.

xiaoqiang-wang, 2021-07-08:
@jinjianrong 
😃  Thank you all the same.

I'm currently learning the amdllpc compiler work-flow, not have a simulator both OK .

sincerely 
xiaoqiang-wang