GPUOpen-Drivers/AMDVLK #241: Prey (2017) - green screen during cut-scene "black-outs"

thomasfsteeples, 2021-07-03:
In the video game Prey (2017), I am experiencing this really annoying visual bug where in cut-scenes where the screen "blacks out", I instead get a screen with green lines. The HUD is preserved, and when the screen actually renders something non-trivial, it's fine.

As an example, I've attached two images from the intro - when Alex has their eyes closed, I get the green screen, but when they go to turn off their alarm clock, rendering kicks in. 

[image]
[image]

Originally posted this to the Proton issues page for the game: https://github.com/ValveSoftware/Proton/issues/4939. Here one of the team members asked me to use a different Vulkan implementation. Using RADV, the black screen renders as expected.

If there any way of getting the desired behaviour under AMDVLK? Any help would be greatly appreciated. Thank you for your time.

jinjianrong, 2021-07-06:
@thomasfsteeples  the issue might have been fixed in LLPC dev branch, could you give it a try?

jinjianrong, 2021-08-02:
Please try 2021.Q3.2 build

closed 2021-08-02 by jinjianrong