GPUOpen-Drivers/AMDVLK #243: Swizzled image view on R8_UNORM is not sampled correctly

llyzs, 2021-07-10:
Steps to reproduce:
1. Create an image with R8_UNORM format
2. Clear the image with color {1.0, 0.0, 0.0, 0.0} using vkCmdClearColorImage
3. (Optional) Render something on the image created above
4. Create an image view with swizzle (VkImageViewCreateInfo.components) equals to {VK_COMPONENT_SWIZZLE_ZERO, VK_COMPONENT_SWIZZLE_ZERO, VK_COMPONENT_SWIZZLE_ZERO, VK_COMPONENT_SWIZZLE_R}. That is, the R channel is being used as Alpha when being sampled.
5. Create an image with B8G8R8A8_UNORM format as framebuffer
6. Render onto the framebuffer by sampling from the image view created in step 3

The result is either complete black or contains many small black blocks (image corruption), and sometimes the driver even freezes.

Note:
* RADV driver does not have the issue.
* It only happens with R8_UNORM format. Use RGBA format with the same swizzle pattern does not have the issue.

JaxLinAMD, 2021-07-12:
looks like a known issue.

llyzs, 2021-07-24:
Confirmed this was fixed with latest release.

closed 2021-07-24 by llyzs