GPUOpen-Drivers/AMDVLK #245: After updating to v-2021.Q3.1 Far Cry 5 is not starting through Lutris

Arjab, 2021-07-20:
I've played Far Cry 5 which I start through Lutris, but after updating amdvlk and lib32-amdvlk to v-2021.Q3.1 the game doesn't even launch. When I downgrade to v-2021.Q2.5 the game starts as expected again. The log Lutris provides is quite uninformative, as you can see here: [log when it works](https://pastebin.com/2qPzQ6Eh), [log when it doesn't work](https://pastebin.com/wdTuRXVR).
[System specifications](https://pastebin.com/jXPqMHTG).

jinjianrong, 2021-07-21:
@Arjab what GPU are you using. Please note support for Pre-Polaris and Pre-Raven asics is dropped starting from 2021.Q2.6

Flakebi, 2021-07-21:
Seems to be a AMD Radeon RX 5700 (navi10) on Arch Linux and KDE (x11) from the log.
It seems to crash somewhere around creating a swapchain, that’s where the non-working log cuts off.
The driver in use looks like radv/aco, but if the bug is fixed by downgrading amdvlk, there seems to be some sideeffect by loading amdvlk.

I guess starting lutris with `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/radeon_icd.x86_64.json` works fine, even when amdvlk 2021.Q3.1 is installed? (That should prevent vulkan from loading the amdvlk driver.)

Arjab, 2021-07-21:
I have no clue what happened, but now everything is back to normal and the game starts with amdvlk v-2021.Q3.1. Closed.

Arjab, 2021-07-24:
Okay, again I have no clue what happened, but now it's not working again. When I uninstall amdvlk and lib32-amdvlk, the game just starts fine.
When I select amd_icd32.json or radeon_icd.i686.json in Lutris, the game's splash screen opens with a dialogue box that just says "Ok" and the Lutris log showing:
"info: Required Vulkan extension VK_KHR_surface not supported
err:   DxvkInstance: Failed to create instance"
When I select either amd_icd64.json or radeon_icd.x86_64.json the game won't start at all.
Currently I have amdvlk 2021.Q3.2-1 and lib32-amdvlk 2021.Q3.1-1 installed.

Arjab, 2021-07-26:
I'm at a point where I don't know whether this has anything to do with amdvlk or if it was pure coincidental. The game works and works not in between reboots without any changes to the system. I'll close this issue again, because I have next to no evidence, that amdvlk is actually the problem and for now it worked again running the latest version.

closed 2021-07-26 by Arjab

Flakebi, 2021-07-26:
Good luck finding the problem!