GPUOpen-Drivers/AMDVLK #246: Missing fragment rasterization in certain case

llyzs, 2021-07-24:
An example to reproduce the issue:
1. Creates a 64x320 image with format B8G8R8A8_UNORM as framebuffer and clear it with green {0.0, 1.0, 0.0, 0.0}
2. Creates another B8G8R8A8_UNORM image of any size as texture and clear it with black {0.0, 0.0, 0.0, 0.0}
3. Creates the following vertex buffer of total 24 vertices (8 triangles), each is of format R32G32B32A32_SFLOAT (x, y, texcoord_x, texcoord_y). Those triangles were designed to draw on the framebuffer from top to bottom:
```
{0,        0,  0.0075,   0},
{0,  0.00625,   0.005,   0},
{2, 0.003125, 0.00625, 0.2},
{0,  0.00625,  0.0075, 0.2},
{0,   0.0125,   0.005, 0.2},
{2, 0.009375, 0.00625, 0.4},
{0,   0.0125,  0.0075, 0.4},
{0,  0.01875,   0.005, 0.4},
{2, 0.015625, 0.00625, 0.6},
{0,  0.01875,  0.0075, 0.6},
{0,    0.025,   0.005, 0.6},
{2, 0.021875, 0.00625, 0.8},
{0,    0.025,  0.0075, 0.8},
{0,  0.03125,   0.005, 0.8},
{2, 0.028125, 0.00625,   1},
{0,  0.03125,  0.0075,   1},
{0,   0.0375,   0.005,   1},
{2, 0.034375, 0.00625, 1.2},
{0,   0.0375,  0.0075, 1.2},
{0,  0.04375,   0.005, 1.2},
{2, 0.040625, 0.00625, 1.4},
{0,  0.04375,  0.0075, 1.4},
{0,     0.05,   0.005, 1.4},
{2, 0.046875, 0.00625, 1.6}
```
4. Vertex shader:
Outputs position (x - 1, 1 - y, 0, 1) and texcoord (texcoord_x / 2, texcoord_y / 2)
```
; SPIR-V
; Version: 1.3
; Generator: Khronos; 0
; Bound: 50
; Schema: 0
               OpCapability Shader
          %1 = OpExtInstImport "GLSL.std.450"
               OpMemoryModel Logical GLSL450
               OpEntryPoint Vertex %2 "main" %10 %12 %23
               OpDecorate %10 Location 0
               OpDecorate %12 Location 1
               OpMemberDecorate %21 0 BuiltIn Position
               OpDecorate %21 Block
          %3 = OpTypeVoid
          %4 = OpTypeFunction %3
          %7 = OpTypeFloat 32
          %8 = OpTypeVector %7 4
          %9 = OpTypePointer Input %8
         %10 = OpVariable %9 Input
         %11 = OpTypePointer Output %8
         %12 = OpVariable %11 Output
         %15 = OpConstant %7 1
         %16 = OpConstant %7 -1
         %17 = OpConstant %7 0
         %18 = OpConstantComposite %8 %15 %16 %17 %17
         %19 = OpConstantComposite %8 %16 %15 %17 %17
         %21 = OpTypeStruct %8
         %22 = OpTypePointer Output %21
         %23 = OpVariable %22 Output
         %24 = OpTypeInt 32 0
         %25 = OpConstant %24 0
         %30 = OpTypePointer Output %7
         %32 = OpConstant %24 1
         %34 = OpConstantComposite %8 %17 %17 %17 %15
         %38 = OpConstant %24 2
         %40 = OpConstant %24 3
         %43 = OpConstant %7 0.5
         %44 = OpConstantComposite %8 %43 %43 %17 %17
          %2 = OpFunction %3 None %4
          %5 = OpLabel
         %13 = OpLoad %8 %10
         %14 = OpVectorShuffle %8 %13 %13 0 1 0 0
         %20 = OpExtInst %8 %1 Fma %14 %18 %19
         %26 = OpAccessChain %11 %23 %25
         %27 = OpCompositeExtract %7 %20 0
         %28 = OpCompositeExtract %7 %20 1
         %29 = OpAccessChain %30 %26 %25
               OpStore %29 %27
         %31 = OpAccessChain %30 %26 %32
               OpStore %31 %28
         %33 = OpAccessChain %11 %23 %25
         %35 = OpCompositeExtract %7 %34 2
         %36 = OpCompositeExtract %7 %34 3
         %37 = OpAccessChain %30 %33 %38
               OpStore %37 %35
         %39 = OpAccessChain %30 %33 %40
               OpStore %39 %36
         %41 = OpLoad %8 %10
         %42 = OpVectorShuffle %8 %41 %41 2 3 2 2
         %45 = OpFMul %8 %42 %44
         %46 = OpCompositeExtract %7 %45 0
         %47 = OpCompositeExtract %7 %45 1
         %48 = OpAccessChain %30 %12 %25
               OpStore %48 %46
         %49 = OpAccessChain %30 %12 %32
               OpStore %49 %47
               OpReturn
               OpFunctionEnd
```
5. Fragment shader:
Just samples using texcoord.
```
; SPIR-V
; Version: 1.3
; Generator: Khronos; 0
; Bound: 27
; Schema: 0
               OpCapability Shader
          %1 = OpExtInstImport "GLSL.std.450"
               OpMemoryModel Logical GLSL450
               OpEntryPoint Fragment %2 "main" %13 %15
               OpExecutionMode %2 OriginUpperLeft
               OpDecorate %9 DescriptorSet 0
               OpDecorate %9 Binding 0
               OpDecorate %13 Location 1
               OpDecorate %15 Location 0
               OpDecorate %18 DescriptorSet 0
               OpDecorate %18 Binding 1
          %3 = OpTypeVoid
          %4 = OpTypeFunction %3
          %7 = OpTypeSampler
          %8 = OpTypePointer UniformConstant %7
          %9 = OpVariable %8 UniformConstant
         %10 = OpTypeFloat 32
         %11 = OpTypeVector %10 4
         %12 = OpTypePointer Input %11
         %13 = OpVariable %12 Input
         %14 = OpTypePointer Output %11
         %15 = OpVariable %14 Output
         %16 = OpTypeImage %10 2D 0 0 0 1 Unknown
         %17 = OpTypePointer UniformConstant %16
         %18 = OpVariable %17 UniformConstant
         %23 = OpTypeVector %10 2
         %24 = OpTypeSampledImage %16
          %2 = OpFunction %3 None %4
          %5 = OpLabel
         %19 = OpLoad %16 %18
         %20 = OpLoad %7 %9
         %21 = OpLoad %11 %13
         %22 = OpVectorShuffle %23 %21 %21 0 1
         %25 = OpSampledImage %24 %19 %20
         %26 = OpImageSampleImplicitLod %11 %25 %22
               OpStore %15 %26
               OpReturn
               OpFunctionEnd
```
6. If call vkCmdDraw(15) to draw 5 triangles (or fewer), it renders correctly, a few lines from top are drawn black.
[image]

7. If call vkCmdDraw(18/21/24) to draw 6/7/8 triangles, the top line of pixels is mysteriously not rendered and left green:
[image]

Tested with RX6700XT driver 2021.Q3.2.

WenqingLiAMD, 2024-06-19:
@llyzs can you test it with latest driver 2024.Q2.2? If the issue persists, can you share the complete source code? Currently I failed to reproduce basing on your description.

WenqingLiAMD, 2024-06-25:
Please feel free to reopen if the issue can be reproduced, many thanks!

closed 2024-06-25 by WenqingLiAMD