GPUOpen-Drivers/AMDVLK #248: amdvlk driver broken after new update

Sydiepus, 2021-08-21:
### setup info

1.  ``lshw -c video`` output :
```
  *-display
       description: VGA compatible controller
       product: HD Graphics 620
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 02
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list
       configuration: driver=i915 latency=0
       resources: irq:130 memory:a1000000-a1ffffff memory:b0000000-bfffffff ioport:5000(size=64) memory:c0000-dffff
  *-display
       description: Display controller
       product: Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445 / 530/535 / 620/625 Mobile]
       vendor: Advanced Micro Devices, Inc. [AMD/ATI]
       physical id: 0
       bus info: pci@0000:01:00.0
       version: c3
       width: 64 bits
       clock: 33MHz
       capabilities: pm pciexpress msi bus_master cap_list rom
       configuration: driver=amdgpu latency=0
       resources: irq:132 memory:90000000-9fffffff memory:a0000000-a01fffff ioport:4000(size=256) memory:a2200000-a223ffff memory:a2240000-a225ffff
```
2. ``lspci -k`` output : 
```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
	Subsystem: Lenovo Device 3817
	Kernel driver in use: skl_uncore
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
	Subsystem: Lenovo Device 39c4
	Kernel driver in use: i915
	Kernel modules: i915
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
	Subsystem: Lenovo Device 3844
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
	Subsystem: Lenovo Device 3836
	Kernel driver in use: intel_pch_thermal
	Kernel modules: intel_pch_thermal
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
	Subsystem: Lenovo Device 383b
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
	Subsystem: Lenovo Device 3843
	Kernel driver in use: mei_me
	Kernel modules: mei_me
00:17.0 RAID bus controller: Intel Corporation 82801 Mobile SATA Controller [RAID mode] (rev 21)
	Subsystem: Lenovo Device 380f
	Kernel driver in use: ahci
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
	Kernel driver in use: pcieport
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
	Kernel driver in use: pcieport
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
	Kernel driver in use: pcieport
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
	Subsystem: Lenovo Device 3817
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
	Subsystem: Lenovo Device 3845
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
	Subsystem: Lenovo Device 380d
	Kernel driver in use: snd_hda_intel
[... 2443 lines cut ...]
	shaderOutputLayer                                  = true
	subgroupBroadcastDynamicId                         = true

VkPhysicalDeviceVulkanMemoryModelFeatures:
------------------------------------------
	vulkanMemoryModel                             = true
	vulkanMemoryModelDeviceScope                  = true
	vulkanMemoryModelAvailabilityVisibilityChains = false

VkPhysicalDeviceWorkgroupMemoryExplicitLayoutFeaturesKHR:
---------------------------------------------------------
	workgroupMemoryExplicitLayout                  = true
	workgroupMemoryExplicitLayoutScalarBlockLayout = true
	workgroupMemoryExplicitLayout8BitAccess        = true
	workgroupMemoryExplicitLayout16BitAccess       = true

VkPhysicalDeviceYcbcrImageArraysFeaturesEXT:
--------------------------------------------
	ycbcrImageArrays = true

VkPhysicalDeviceZeroInitializeWorkgroupMemoryFeaturesKHR:
---------------------------------------------------------
	shaderZeroInitializeWorkgroupMemory = true

```

Sydiepus, 2021-08-21:
### Additional info.
- ```env AMD_VULKAN_ICD=AMDVLK vkcube``` output for ```amdvlk 2021.Q3.4-1``` :
```
vkEnumeratePhysicalDevices reported zero accessible devices.

Do you have a compatible Vulkan installable client driver (ICD) installed?
Please look at the Getting Started guide for additional information.
```
- ```amdvlk 2021.Q3.2-1``` is working.
-  there is no ```amdvlk 2021.Q3.3-1``` in [Arch linux archive](https://archive.archlinux.org/packages/a/amdvlk/)

Johnnynator, 2021-08-21:
I assume you only tested `env AMD_VULKAN_ICD=RADV vulkaninfo` while the old amdvlk was installed?

Since `AMD_VULKAN_ICD` env is currently not enabled, because the implicit_layer that was providing it was broken-ish if both mesa and amdvlk were installed. (3.4 should fix it, will take a look at it).
Right now you can switch between radv and amdvlk with e.g. `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd64.json` (or `radeon_icd.x86_64.json` for radv)

Furthermore they removed support for your GPU (or like all Pre-Polaris and Pre-Raven GPUs) in 2021.Q2.6.

On another note, do you have VK_ICD_FILENAMES already set to smth? Since I think vk would usually default to radv and not amdvlk.

Sydiepus, 2021-08-23:
I tested `env AMD_VULKAN_ICD=RADV vulkaninfo`
With both the new and the old one installed. View the comment additional info.   

No VK_ICD_FILENAMES is not set to anything.   
I tried setting :
 ```
VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd64.json
```    
But i was out of luck as dxvk was giving an error about ``` not finding an adapter check filter list```

Something like that. 

even when using :  ```radeon_icd.x86_64.json```

Sydiepus, 2022-04-09:
Closing issue since support for my GPU was dropped and the issue was solved by using mesa.

closed 2022-04-09 by Sydiepus