GPUOpen-Drivers/AMDVLK #250: Doom Eternal crash at start with AMD-Pro 21.20 and AMD-Pro 21.30

pingubot, 2021-08-29:
Hi,

as the title says, Doom Eternal is crashing with the 2 mentioned pro drivers at start.
AMD-Pro 21.10 works and also amdvlk versions up to 2021q34 work. Also Radv has no issues.

My system:

```
System:    Host: gamebox Kernel: 5.13.12-tkg-idlefix x86_64 bits: 64 compiler: gcc v: 11.1.1 Desktop: KDE Plasma 5.22.4 
           tk: Qt 5.15.2 wm: kwin_x11 dm: SDDM Distro: openSUSE Tumbleweed 20210826 
CPU:       Info: 8-Core model: AMD Ryzen 7 3700X bits: 64 type: MT MCP arch: Zen 2 rev: 0 cache: L2: 4 MiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 115093 
           Speed: 1965 MHz min/max: 2200/3600 MHz boost: enabled Core speeds (MHz): 1: 1965 2: 2205 3: 2196 4: 2191 5: 2196 
           6: 2195 7: 2195 8: 2195 9: 2195 10: 2195 11: 2195 12: 2195 13: 2196 14: 2085 15: 1862 16: 1860 
Graphics:  Device-1: Advanced Micro Devices [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] driver: amdgpu v: kernel 
           bus-ID: 0a:00.0 chip-ID: 1002:73bf 
           Display: x11 server: X.Org 1.20.13 compositor: kwin_x11 driver: loaded: amdgpu resolution: 3440x1440~120Hz 
           s-dpi: 96 
           OpenGL: renderer: AMD Radeon RX 6900 XT (SIENNA_CICHLID DRM 3.41.0 5.13.12-tkg-idlefix LLVM 12.0.1) 
           v: 4.6 Mesa 21.2.0 direct render: Yes 
```

Many thx,
Christian

perlfu, 2021-08-30:
I believe I can reproduce this issue on a Navi21.
I had to remove the switchable graphics layer as this prevented Proton Experimental from detecting a GPU.
The crash happens after the intro movies/logos, just after login to Bethesda.net ?

jinjianrong, 2021-09-01:
The issue is fixed in internal 21.40 build

pingubot, 2021-09-02:
> The issue is fixed in internal 21.40 build

Cool, thx, any eta when that may show up ?

jinjianrong, 2021-09-15:
it's mid Nov

pingubot, 2021-11-13:
Can confirm it works again with Pro 21.40 !

closed 2021-11-16 by jinjianrong