GPUOpen-Drivers/AMDVLK #251: vk_ext_transform_feedback & VK_EXT_SAMPLER_FILTER_MINMAX broken on GCN POLARIS

RinMaru, 2021-08-31:
[image]

VK_EXT_SAMPLER_FILTER_MINMAX broken on Windows but working on LINUX open source drivers

[image]

Transform Feedback broken

jinjianrong, 2021-09-09:
@RinMaru could you provide a Renderdoc trace for us to reproduce the issue? could you provide two images to show the issue: one with VK_EXT_SAMPLER_FILTER_MINMAX broken and one with VK_EXT_SAMPLER_FILTER_MINMAX working?

RinMaru, 2021-09-09:
> @RinMaru could you provide a Renderdoc trace for us to reproduce the issue? could you provide two images to show the issue: one with VK_EXT_SAMPLER_FILTER_MINMAX broken and one with VK_EXT_SAMPLER_FILTER_MINMAX working?

RinMaru, 2021-09-09:
I currently have no way to do that I just know it's broken I can try to see if theres another way to repro.

RinMaru, 2021-09-09:
Since I have no clue of anything that uses the extension outside an emulator I can't test. If there's a tool to test the offending extension then I'll use it Otherwise I guess I'll close this

perlfu, 2021-09-09:
@RinMaru which emulator do you see the issue on? Yuzu?

RinMaru, 2021-09-09:
> @RinMaru which emulator do you see the issue on? Yuzu?

Yes. Since then the sampler ext has been disabled as a workaround. Transform feedback is still an issue though.

FernandoS27, 2021-09-11:
@jinjianrong Hi I'm a yuzu developer. We have prepared a test example https://drive.google.com/file/d/1oTUR3evy49TiBYC2dpNyRMSPhZCE7m89/view?usp=sharing

it has a renderdoc capture done on windows(since you guys share some codebase) and  a homebrew we wrote. You can test it with our emulator: https://github.com/yuzu-emu/yuzu-mainline/releases

If you want I can send you an Intel/NVIDIA example so you have a reference solution.

About VK_EXT_SAMPLER_FILTER_MINMAX. We use nouveau and deko3d to produce homebrew for Switch, I don't think I have the option there for this feature and I can't provide a capture from a game as that contains copyright material. In general, it's broken in GCN 4, 3, 2, 1 on Windows proprietary driver and RADV/Mesa

FernandoS27, 2021-09-11:
As a side note, I need to debug a TDR on AMD GPUs. It happens on every GPU, every driver. I tried Radeon Profiler but didn't help, What's the best way to debug this?

Flakebi, 2021-09-14:
> I need to debug a TDR on AMD GPUs.

[umr](https://gitlab.freedesktop.org/tomstdenis/umr) can be used for debugging (ssh into the hanging machine and run umr).
`umr -t` gives an overview over the hardware blocks, although only with cryptic acronyms.

If a shader is hanging, `umr -O bits,disasm_anyways --waves gfx_0.0.0 --gfxoff 0` can be used to show the currently executed instruction. In the case of a segfault, it usually hangs at a `s_waitcnt`.

People in the `#dri-devel` irc channel on freenode can probably help more with debugging hangs (on linux).

jinjianrong, 2021-09-16:
@FernandoS27 what GPU do you use to capture the renderdoc trace? what issue are you seeing with the trace? This is not for the VK_EXT_SAMPLER_FILTER_MINMAX issue, right?

FernandoS27, 2021-09-16:
> @FernandoS27 what GPU do you use to capture the renderdoc trace? what issue are you seeing with the trace? This is not for the VK_EXT_SAMPLER_FILTER_MINMAX issue, right?

I used an RX 6800M - windows 10, I can also provide Vega and GCN(RX 570) Captures if needed as well as Linux captures. In general, transform feedback doesn't seem to write anything. The capture is a particle emitter, it should generate a bunch of falling particles.

about the VK_EXT_SAMPLER_FILTER_MINMAX , I'll see how I can make a test. I may have to make one outside of emu (meaning writing all the Vulkan boilerplate code *sight*). I'll make an issue myself when I have a reproducible test for that.

FernandoS27, 2021-09-16:
Here#s a capture in RADV (Mesa) where Transform feedback works: https://drive.google.com/file/d/1ydXnxOwQJ0wDNefyWr_AIhjeCqsgINYn/view?usp=sharing

Same GPU: RX 6800M, Ubuntu 21.04

jinjianrong, 2021-09-17:
@FernandoS27  we are seeing   VK_ERROR_OUT_OF_HOST_MEMORY error message when open the renderdoc trace in https://drive.google.com/file/d/1oTUR3evy49TiBYC2dpNyRMSPhZCE7m89/view?usp=sharing.  I assume this is the issue mentioned in https://github.com/GPUOpen-Drivers/AMDVLK/issues/253. Could you provide a renderdoc trace without this issue? Also what version of Windows driver are you using?

FernandoS27, 2021-09-17:
> @FernandoS27 we are seeing VK_ERROR_OUT_OF_HOST_MEMORY error message when open the renderdoc trace in https://drive.google.com/file/d/1oTUR3evy49TiBYC2dpNyRMSPhZCE7m89/view?usp=sharing. I assume [...]

Here you go https://drive.google.com/file/d/17x_8cyaJuzRuyw9u-WVoB3WeQiMB4wJY/view?usp=sharing Same OS: windows, Same GPU RX 6800M

RinMaru, 2021-10-05:
This is still broke in Newest Driver 21.10.1

FernandoS27, 2021-12-16:
@jinjianrong It seems to be fixed on latest amdvlk version. I still need further confirmation.

FernandoS27, 2021-12-16:
[image]
I can confirm now.

jinjianrong, 2021-12-16:
Thanks for the confirm

closed 2021-12-16 by jinjianrong

FernandoS27, 2022-01-20:
@jinjianrong as far as I know Windows AMD proprietary driver uses AMDVLK, right? Is there any way to know which version of it is used? January's Windows driver releases have not included this fix yet. So I'm not sure if AMDVLK has not been updated there or the bug happens on windows for some reason.

jinjianrong, 2022-01-27:
@FernandoS27 please try this Windows driver release: https://www.amd.com/en/support/kb/release-notes/rn-rad-win-22-1-2-vulkan-1-3

jinjianrong, 2022-01-27:
Btw, you could use vulkaninfo to get driverVersion (refer to https://github.com/GPUOpen-Drivers/AMDVLK/issues/262)  to check if a Windows driver and an AMDVLK driver are built from similar code base.

FernandoS27, 2022-01-27:
@jinjianrong i have checked up that driver and transform feedback is still broken in Windows. I had checked Linux about 2 weeks ago and TFB was still working.

This is our own report from Vulkan VkPhysicalDeviceProperties
```
[  10.479230] Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:186: Driver: AMD proprietary driver 2.0.213
[  10.479233] Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:187: Driver Version: 2.0.213
[  10.479234] Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:188: Device: Radeon RX 570 Series
[  10.479235] Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:189: Vulkan: 1.3.203
```

Not single TFB test passed.