GPUOpen-Drivers/AMDVLK #253: VkPhysicalDeviceFloat16Int8FeaturesKHR on create physical device throws VK_ERROR_OUT_OF_HOST_MEMORY

FernandoS27, 2021-09-13:
It seems to be regression as this worked on the windows driver 21.8.2, I just experienced it in 21.9.1. AMDVLk users had reported before hand that it was causing a crash

Moonlacer, 2021-09-13:
I second this, happens in my testing as well

RinMaru, 2021-09-15:
[Yuzu mainline 747](https://github.com/yuzu-emu/yuzu-mainline/releases/tag/mainline-0-747) Link so you dont have to search the Repo

jinjianrong, 2021-09-16:
@FernandoS27 we are not allowed to run the emulator. could the issue be reproduced with the renderdoc capture in https://github.com/GPUOpen-Drivers/AMDVLK/issues/251?

FernandoS27, 2021-09-16:
> @FernandoS27 we are not allowed to run the emulator. could the issue be reproduced with the renderdoc capture in #251?

Should be possible as I captured it before patching it.

FernandoS27, 2021-09-16:
After rechecking. It turned out to be an issue in our side that was shadowed and didn't hppen in other drivers for some reason. 8memory coprruption).

closed 2021-09-16 by FernandoS27