GPUOpen-Drivers/AMDVLK #255: Requesting for Vulkan Video extensions support

nyanmisaka, 2021-11-06:
The [Vulkan video standard](https://www.khronos.org/blog/an-introduction-to-vulkan-video) was published in provisional form in April 2021. And the proprietary Vulkan video extension have been implemented in AMDGPU-PRO driver for years and it's [only used in AMD AMF internally](https://github.com/GPUOpen-LibrariesAndSDKs/AMF/issues/4#issuecomment-645550088).

Any chance we can get some beta drivers on AMDVLK for Vulkan video support?

NVIDIA has released [beta Vulkan drivers](https://developer.nvidia.com/vulkan-driver) that implement Vulkan Video and a [decoding sample](https://github.com/nvpro-samples/vk_video_samples) was also published for developers. Hope we will not fall behind.

Extensions for Vulkan video:
```
VK_KHR_video_queue
VK_KHR_video_decode_queue
VK_KHR_video_encode_queue

VK_EXT_video_decode_h264
VK_EXT_video_encode_h264
VK_EXT_video_decode_h265
VK_EXT_video_encode_h265
```

[image]

oscarbg, 2023-05-21:
Any update on that? Radv has support already.. Seems amd released a windows beta driver with support in november 2022 so half a year ago.. also would be nice if vulkan video encoding beta support could be shipped also similar to nvidia..

oscarbg, 2023-06-30:
well at least now we have a Windows "stable driver" seems with Vulkan video: "Adrenalin Edition 23.10.01.41 for Additional Vulkan Extensions Release Notes"
hope coming soon to amdvlk..
seems also new exts like [VK_KHR_cooperative_matrix](https://www.khronos.org/registry/vulkan/specs/1.2-extensions/man/html/VK_KHR_cooperative_matrix.html) are supported..