GPUOpen-Drivers/AMDVLK #256: Strange visual corruption when rasterizing to a VkImage that was previously used in a compute shader

haasn, 2021-11-06:
See https://github.com/mpv-player/mpv/issues/9389 for context. But basically, I get artifacts like these:

[image]

This is with blending enabled, but that part seems to be working correctly (though the alpha channel seems to be derived from the same sort of corruption as the rgb color values). I get corruption even with a trivial fragment shader like `void main() { out_color = vec4(0.2, 0.4, 0.6, 1.0);`, in which case it looks more like this:

[image]

This is very much consistent with the color values output by the shader themselves being somehow horrifically mangled/misinterpreted by the hardware. This is very surprising given that it's the rasterization output of a fragment shader, rather than being read anywhere from memory.

The only consistent reproducing circumstance seems to be:

1. Render to a VkImage using `imageStore` from a compute shader.
2. Render to that same VkImage using a fragment shader.

Notably, any other operation involving the VkImage seems to work fine. I can download it again and it will look correct on the host. I can sample from it in other shaders, even other fragment shaders. It's only *rendering* to them that seems to break, and only from fragment shaders.

haasn, 2021-11-06:
I also managed to get corruption like this, by rendering to an image and then blitting (vkCmdCopyImage) it to a different image:

[image]

I really have no idea what's going here anymore, other than that it's very broken and only seems to affect AMDVLK.

haasn, 2021-11-06:
Update: This is a regression according to [this comment](https://github.com/mpv-player/mpv/issues/9389#issuecomment-962517250)

jinjianrong, 2022-11-21:
The issue is fixed in 2022.Q3.1 and later release

closed 2022-11-21 by jinjianrong