GPUOpen-Drivers/AMDVLK #262: Please provide driver version number in driverInfo field

smcv, 2021-12-06:
While working on Steam's steam-runtime-system-info diagnostic tool, I've been looking into getting better version reporting for Vulkan drivers.

The problem is that the `driverVersion` in [VkPhysicalDeviceProperties](https://www.khronos.org/registry/vulkan/specs/1.2-extensions/man/html/VkPhysicalDeviceProperties.html) is a single integer with no standardized encoding. Mesa uses the same encoding as for the `apiVersion`:

* Bits 29-31 (3  bits, mask with 0xe000_0000): variant
* Bits 22-28 (7  bits, mask with 0x1fc0_0000): major
* Bits 12-21 (10 bits, mask with 0x003f_f000): minor
* Bits  0-11 (12 bits, mask with 0x0000_0fff): patch (micro)

but unfortunately the Nvidia proprietary driver seems to have its own encoding:

* Bits 22-31 (10 bits, mask with 0xffc0_0000): major
* Bits 14-21 ( 8 bits, mask with 0x003f_c000): minor
* Bits 6-13 ( 8 bits, mask with 0x0000_3fc0): micro
* Bits 0-5 ( 6 bits, mask with 0x0000_003f): nano (probably not used in public drivers?)

AMDVLK 2021.Q4.1 reports its version number as 0x8000cb. If it's the same encoding as an `apiVersion`, that would be 2.0.203 - is that a meaningful version number for this driver? It doesn't seem to have any obvious relationship with the public-facing version number 2021.Q4.1.

The convention seems to be that the `driverInfo` in [VkPhysicalDeviceVulkan12Properties](https://www.khronos.org/registry/vulkan/specs/1.2-extensions/man/html/VkPhysicalDeviceVulkan12Properties.html) is a human-readable version number of some sort. For example, for Mesa drivers like RADV and Intel ANV it's something like `Mesa 21.3.0`, and for Nvidia proprietary it's something like `460.91.03`, in both cases matching the user-facing version numbers used in downloadable binaries.

Would it be possible to fill this field with whatever version number you would want your users to give when describing this driver? For example it could be something like one of these:

* `2021.Q4.1`
* `2021.Q4.1 (2.0.203)`
* `2021.Q4.1 (LLVM 13.0.1, XGL 1.2.3, LLPC 4.5, PAL 7.8.9)`
* `2.0.203`

JaxLinAMD, 2021-12-16:
We decided to fill release version ( e.g. "2021.Q4.2“) into driverInfo in VkPhysicalDeviceVulkan12Properties since next release.(Q4.3)
The driverVersion in VkPhysicalDeviceProperties is an internal version, for more details see https://github.com/GPUOpen-Drivers/amd-vulkan-versions

smcv, 2021-12-16:
> We decided to fill release version ( e.g. "2021.Q4.2“) into driverInfo

Thanks, that's what I hoped for.

Steam's diagnostic tool in the public beta version should display this correctly. That change is not yet in the stable version, but will arrive next time the beta changes get merged.

JaxLinAMD, 2021-12-17:
to clarify, this decision  is only for amdvlk driver ( linux only)

smcv, 2021-12-17:
That's fine, steam-runtime-system-info is also Linux-specific.

If AMD has Windows Vulkan drivers, then those should probably also populate `driverInfo` with something users will understand (perhaps they already do, I don't know), but I'm personally only interested in Linux.

WenqingLiAMD, 2022-08-05:
Driver info updated to this form: `2022.Q3.2 (LLPC)`

smcv, 2022-08-05:
> Driver info updated to this form: `2022.Q3.2 (LLPC)`

We just display this string, and we don't parse or interpret it, so the specific format is not important: any string that will be understood by your users and developers is equally good.

Current versions of Steam should display this in Help -> System Information: wait for the detailed report to be generated, then look for "x11/vulkan". Sorry, I don't have an AMDVLK system set up at the moment, so I'm not able to verify this.

closed 2022-08-08 by WenqingLiAMD