GPUOpen-Drivers/AMDVLK #265: Invalid pointer crash

Chryseus, 2021-12-28:
Running any application using AMDVLK including vulkaninfo results in a crash, `free(): invalid pointer`, RADV works just fine.
This appears to have been caused by last nights update to 2021.Q4.3-1 from 4.2-1.
I'm running Arch Linux 5.15.11-arch2-1
[glxinfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/7784298/glxinfo.txt)

cycomanic, 2021-12-28:
I encounter the same issue. vulkan-radeon works, but amdvlk crashes. 

```
$vulkaninfo --summary
free(): invalid pointer
fish: Job 1, 'vulkaninfo --summary' terminated by signal SIGABRT (Abort)
```

Relevant section from journal:
```

Dec 28 11:59:30 endeavourBase audit[16799]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=3 pid=16799 comm="vulkaninfo" exe="/usr/bin/vulkaninfo" sig=6 res=1
Dec 28 11:59:30 endeavourBase kernel: audit: type=1701 audit(1640689170.546:147): auid=1000 uid=1000 gid=1000 ses=3 pid=16799 comm="vulkaninfo" exe="/usr/bin/vulkaninfo" si>
Dec 28 11:59:30 endeavourBase audit: BPF prog-id=38 op=LOAD
Dec 28 11:59:30 endeavourBase audit: BPF prog-id=39 op=LOAD
Dec 28 11:59:30 endeavourBase audit: BPF prog-id=40 op=LOAD
Dec 28 11:59:30 endeavourBase kernel: audit: type=1334 audit(1640689170.551:148): prog-id=38 op=LOAD
Dec 28 11:59:30 endeavourBase kernel: audit: type=1334 audit(1640689170.551:149): prog-id=39 op=LOAD
Dec 28 11:59:30 endeavourBase kernel: audit: type=1334 audit(1640689170.551:150): prog-id=40 op=LOAD
Dec 28 11:59:30 endeavourBase systemd[1]: Started Process Core Dump (PID 16801/UID 0).
░░ Subject: A start job for unit systemd-coredump@2-16801-0.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit systemd-coredump@2-16801-0.service has finished successfully.
░░ 
░░ The job identifier is 2499.
Dec 28 11:59:30 endeavourBase audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@2-16801-0 comm="systemd" exe="/usr/lib/systemd/s>
Dec 28 11:59:30 endeavourBase kernel: audit: type=1130 audit(1640689170.552:151): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@2-16801-0 comm="syst>
Dec 28 11:59:30 endeavourBase systemd-coredump[16802]: [🡕] Process 16799 (vulkaninfo) of user 1000 dumped core.
                                                       
                                                       Module linux-vdso.so.1 with build-id e03e20cb310dd7f3aa2cff60ad39737026d9ebe6
                                                       Module libxcb-dri2.so.0 with build-id 2dd6e65129a809dab828a1d26215a3f7a363fcc8
                                                       Module libnss_files.so.2 with build-id 1a36dfc01d3a1010b2ee79766a24a8090a3266d5
                                                       Module libcrypto.so.1.1 with build-id 4c926b672d97886b123e03a008387aecf0786de4
                                                       Module libssl.so with build-id 1024424ab33a3767da03f4fdb1fc1b02479f160f
                                                       Module amdvlk64.so with build-id 45a23aea5cfa127688d1c3f041b9f9d570f748c1
                                                       Module libicudata.so.70 with build-id e1dcc2a88cfaafed882d09c90c668af0eed4efed
                                                       Module liblzma.so.5 with build-id 8b615460aa230708c5183f16bede67aa0437d95e
                                                       Module libicuuc.so.70 with build-id 2e245c2bf12f95fd8ab79b3a4be99524677cbd70
                                                       Module libxml2.so.2 with build-id b710b880b0d4e9c93554fa632129db44c20632cd
                                                       Module libncursesw.so.6 with build-id 859adb8b03a5655151b4da104bc4b326a0d4bc8c
                                                       Module libedit.so.0 with build-id 4b0babfcad161c2ad0af6e59e2493258db23a331
                                                       Module libdrm.so.2 with build-id 457a1a1bd25f1d5541a10d3f16c5dc1440d77c7d
                                                       Module libxshmfence.so.1 with build-id 8876d9ccf620858795724ca24b9e567585a77cec
                                                       Module libxcb-shm.so.0 with build-id fb797f299a446f559a95afcc168227482cc800d1
                                                       Module libxcb-randr.so.0 with build-id b6c1c7d4b28863b55ade251813db7f246b7ffe8b
                                                       Module libxcb-sync.so.1 with build-id dda14591103b01b1311906053bf1ca9e82ade35c
                                                       Module libxcb-present.so.0 with build-id 68f5465258750e2397b1333b3ffc01ee33caa4e1
                                                       Module libX11-xcb.so.1 with build-id 0db4f94d8ae31b8dc9a83f825a9171656f1e532c
                                                       Module libzstd.so.1 with build-id 4b10444c1560ebc574af4d5f488b7408b22d450e
                                                       Module libexpat.so.1 with build-id 16023a53e7b5bbdff92bd0bf2eebd36f9b5b879f
                                                       Module libz.so.1 with build-id 81bf6e728a6d6f5b105b0f8b25f6c614ce10452a
                                                       Module libxcb-dri3.so.0 with build-id 9407a2480e09dc5a1dd9d9a0652fa8d32b328c91
                                                       Module libelf.so.1 with build-id 4300a0a24ac790c27ac0fe1fd5253729905ce708
                                                       Module libdrm_amdgpu.so.1 with build-id 30d019987d514b23115b47d6b47c1e45c18fdbaf
                                                       Module libLLVM-13.so with build-id 82982c299d46219d0eab8ee9cb8bb7365483453b
[... cut ...]
░░ should be reported to its vendor as a bug.
Dec 28 11:59:30 endeavourBase systemd[1]: systemd-coredump@2-16801-0.service: Deactivated successfully.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit systemd-coredump@2-16801-0.service has successfully entered the 'dead' state.
Dec 28 11:59:30 endeavourBase audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@2-16801-0 comm="systemd" exe="/usr/lib/systemd/sy>
```

ahcm, 2021-12-29:
Also breaks Bannerlord, downgrading to amdvlk-2021.Q4.2 fixes it.

rj1, 2021-12-29:
can reproduce. with this version upgrade on arch linux, path of exile no longer launches w/ proton/steam. same error  `free(): invalid pointer`

PhavourFlavour, 2021-12-30:
I can confirm more games are affected
- Days Gone
- Zero Horzon Dawn

Downgrading mentionend in first post works.

lukedupin, 2021-12-31:
I can confirm all of my wine games are broken with Q4.3

jivvy, 2021-12-31:
Godot game engine is also broken with Q4.3

lordheavy, 2021-12-31:
We have issues with lto. It should be fixed with 2021.Q4.3-2 package.

Chryseus, 2021-12-31:
2021.Q4.3-2 seems to have fixed the issue.

ahcm, 2021-12-31:
2021.Q4.3-2 fixed it.

closed 2021-12-31 by Chryseus