GPUOpen-Drivers/AMDVLK #266: Memory leak in Far Cry 6

DoomPenguin9, 2022-01-02:
There is a memory leak in Far Cry 6. By the time I load into the map and I can actually play the game, pretty much all of my RAM is full, and the performance is extremely poor as a consequence of that.
[image]

This does not happen with RADV. Screenshot with RADV for comparison:

[image]

System:
OS: Arch Linux
Kernel: 5.15.12-zen1-1-zen
CPU: Ryzen 7 5800X
GPU: RX 6700XT
AMDVLK version: 2021.Q4.3
WINE Version: wine-tkg-staging-fsync-git 7.0rc3.r1.g440e5eee-326
VKD3D version: vkd3d-proton(git-c0a3fa8)

RicArch97, 2022-01-19:
Probably the same issue for me, using AMDVLK 2022.Q1.1;

In the loading screen of Horizon Zero Dawn where the game is presumably pre-compiling shaders, RAM and some swap filled up completely in seconds (I have 32GB RAM), and the game crashed after a few seconds.

Same behaviour in overwatch, during the pre-compiling fase, game crashed.

Not to mention that the pre-compiling takes ages on my AMD Ryzen 9 5950X, which has 32 logical cores, with only like 20-30% of those being used as well.

Toorero, 2022-01-28:
I have pretty much the same issue as @RicArch97 but in Monster Hunter World with.
32 GB RAM full in seconds during pre-compilation in the loading-screen. The compilation is pretty quick though but always results in a live lock of the system before completion.

---
Now a comparison of RAM usage. The game was started at 40s and killed in the first graph, so I don't run into a live lock situation.

1. AMDVLK:
[image]
2. Mesa:
[image]
---
##System:##
- GPU: 5700XT
- AMDVLK version: AMDVLK 2022.Q1.1
- OS: Arch Linux
- Kernel: 5.16.2-zen1-1-zen
- Proton version: 1:7.0rc6.GE.1-1

pingubot, 2022-06-05:
I guess https://github.com/GPUOpen-Drivers/AMDVLK/issues/230 is basically the same problem.

closed 2024-05-22 by jinjianrong