GPUOpen-Drivers/AMDVLK #267: Rendering issue in Battlefield 1 due to anisotropic filtering for single level images

ms178, 2022-01-06:
RADV and AMDVLK both now make use of anisotropic filtering for single level images according to a Mesa developer, leading to very noticeable texture rendering issue in Battlefield 1. I first reported this issue to Mesa right after their implementation landed in Mesa-git (see https://gitlab.freedesktop.org/mesa/mesa/-/issues/5753 with more screenshots and the comment from the Mesa dev), but during testing I found that AMDVLK is also affected. I tried DX11 and DX12 (via DXVK and VKD3D-Proton), both render pathes show the issue.

The Mesa dev thinks it is a bug of the game and works on a workaround. I hope it can also be worked around in AMDVLK.

[image]

My GPU is a Vega 56.

ms178, 2022-09-15:
This still reproduces with the current AMDVLK. Is there an ETA for a fix?

WenqingLiAMD, 2022-11-16:
> This still reproduces with the current AMDVLK. Is there an ETA for a fix?

Could you help verify it with AMDVLK v-2022.Q4.2?

ms178, 2022-11-16:
@WenqingLiAMD I'd love to test that, but Battlefield 1 is broken right now on Linux due to an incompatibility between the new EA web app and Proton. Nevertheless, as the new AMDVLK release notes mentioned a fixed rendering issue in Battlefield 1, I thought that it meant this issue, but I will check that to make sure, once the mentioned issue with the EA web app is resolved.

closed 2022-11-24 by jinjianrong