GPUOpen-Drivers/AMDVLK #272: Crashes on artix with

ccuntd, 2022-02-01:
vulkaninfo and other programs crash with:
```
$ vulkaninfo
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Driver /usr/lib/libvulkan_radeon.so supports Vulkan 1.2, but only supports loader interface version 4. Interface version 5 or newer required to support this version of Vulkan (Policy #LDP_DRIVER_7)
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Driver /usr/lib/amdvlk64.so says it supports interface version 6 but still exports core entrypoints (Policy #LDP_DRIVER_6)
ERROR: Incompatible compiler options cross compiler instances!*** stack smashing detected ***: terminated
Aborted
```

here's my gpu info using vulkaninfo with mesa's vulkan driver (--summary)
```
Devices:
========
GPU0:
	apiVersion         = 4202691 (1.2.195)
	driverVersion      = 88092677 (0x5403005)
	vendorID           = 0x1002
	deviceID           = 0x1638
	deviceType         = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
	deviceName         = AMD RADV RENOIR
	driverID           = DRIVER_ID_MESA_RADV
	driverName         = radv
	driverInfo         = Mesa 21.3.5
	conformanceVersion = 1.2.3.0
GPU1:
	apiVersion         = 4202691 (1.2.195)
	driverVersion      = 88092677 (0x5403005)
	vendorID           = 0x1002
	deviceID           = 0x73ff
	deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
	deviceName         = AMD RADV DIMGREY_CAVEFISH
	driverID           = DRIVER_ID_MESA_RADV
	driverName         = radv
	driverInfo         = Mesa 21.3.5
	conformanceVersion = 1.2.3.0
```

ccuntd, 2022-02-01:
ahh, i don't this title, can't come up with something better tho

Flakebi, 2022-02-01:
I can reproduce this on a stock arch linux, even without GPU.
Dockerfile to reproduce:
```Dockerfile
From archlinux
RUN pacman -Sy --noconfirm archlinux-keyring
RUN pacman -Su --noconfirm xterm amdvlk libffi vulkan-tools vulkan-icd-loader
CMD vulkaninfo
```
To run, save as `Dockerfile` in an empty folder and execute `docker build -t arch . && docker run --rm arch`.
Output:
```
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Driver /usr/lib/amdvlk64.so says it supports interface version 6 but still exports core entrypoints (Policy #LDP_DRIVER_6)
ERROR at /build/vulkan-tools/src/Vulkan-Tools-1.2.203/vulkaninfo/vulkaninfo.h:672:vkCreateInstance failed with ERROR_OUT_OF_HOST_MEMORY
```

I suppose this happens because arch has a new vulkan loader (1.2.203). On another linux distribution with 1.2.198, vulkaninfo works just fine with amdvlk 2022.Q1.2.

ccuntd, 2022-02-15:
but it said apiVersion         = 4202691 (1.2.195)
is that the vulkan loader because it sounds close to 1.2.198

Flakebi, 2022-02-17:
Oh, I misinterpreted the error. The crucial part actually is `Incompatible compiler options cross compiler instances!`. This has been fixed internally and will be part of one of the next two releases.

CrezyDud, 2022-02-28:
nice

> Oh, I misinterpreted the error. The crucial part actually is `Incompatible compiler options cross compiler instances!`. This has been fixed internally and will be part of one of the next two releases.

CrezyDud, 2022-02-28:
```
Devices:
========
GPU0:
	apiVersion         = 4206797 (1.3.205)
	driverVersion      = 8388824 (0x8000d8)
	vendorID           = 0x1002
	deviceID           = 0x687f
	deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
	deviceName         = AMD Radeon RX Vega
	driverID           = DRIVER_ID_AMD_OPEN_SOURCE
	driverName         = AMD open-source driver
	driverInfo         = 2022.Q1.3
	conformanceVersion = 1.3.0.0
```
WORKS!
we can close this now!

Flakebi, 2022-04-06:
The issue with multiple GPUs should be fixed in the [2022.Q2.1](https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2022.Q2.1) release

closed 2022-08-05 by WenqingLiAMD