GPUOpen-Drivers/AMDVLK #278: err:vulkan:wine_vkCreateInstance Failed to load physical devices, res=5

raupach, 2022-06-22:
The following error occurs with version v-2022.Q2.3 and the game crashes (the error is independent of any game and occurs with all of them):

```
Error Log:
[cli] INFO: Logging in...
[Core] INFO: Trying to re-use existing login session...
[cli] INFO: Checking for updates...
[Core] INFO: Getting authentication token...
[cli] INFO: Launching 8e425d507bd6473dbb927f0fea8ef4f6...
0024:fixme:shell:InitNetworkAddressControl stub
0024:fixme:imm:ImeSetActiveContext (007DDAA0, 1): stub
0024:fixme:imm:ImmReleaseContext (0001005C, 007DDAA0): stub
0074:fixme:imm:ImeSetActiveContext (0000000000354AE0, 0): stub
0074:fixme:imm:ImmReleaseContext (0000000000010020, 0000000000354AE0): stub
00fc:fixme:imm:ImeSetActiveContext (048F2B10, 1): stub
00fc:fixme:imm:ImmReleaseContext (0002004E, 048F2B10): stub
00fc:fixme:msvcp:_Locinfo__Locinfo_ctor_cat_cstr (047CF33C 1 C) semi-stub
00fc:fixme:msvcp:_Locinfo__Locinfo_ctor_cat_cstr (047CF0F4 1 C) semi-stub
00fc:err:vulkan:wine_vkCreateInstance Failed to load physical devices, res=5
00fc:fixme:msvcrt:__clean_type_info_names_internal (09C4A67C) stub
00fc:fixme:msvcrt:__clean_type_info_names_internal (003D85BC) stub
00fc:fixme:msvcrt:__clean_type_info_names_internal (050481E4) stub
00fc:fixme:msvcrt:__clean_type_info_names_internal (04F24188) stub
00fc:fixme:msvcrt:__clean_type_info_names_internal (105ED210) stub
```

Environment:
wine 7.11
heroic version 2.3.9
linux arch

When I roll back the version to v-2022.Q2.2 the error does not occur.

tycho, 2022-06-22:
I've noticed the same issue. It seems that `vkEnumeratePhysicalDevices` returns `VK_INCOMPLETE` even when provided with an appropriately sized array of `VkPhysicalDevice`s (as determined by a previous call to get the size required), and apps are getting confused by this, with good reason. Even something as simple as `vkcube` or `vulkaninfo` will break.

mihalycsaba, 2022-06-23:
Had the same problem on arch, had to rollback

jinjianrong, 2022-06-24:
@raupach @tycho @mihalycsaba what version of Vulkan loader is installed in your system?

mihalycsaba, 2022-06-24:
> @raupach @tycho @mihalycsaba what version of Vulkan loader is installed in your system?

```
pacman -Qi vulkan-icd-loader
Name            : vulkan-icd-loader
Version         : 1.3.217-1
```

mihalycsaba, 2022-06-24:
Looked around in the mentioned issues, adding VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/radeon_icd.x86_64.json fixes the problem.

konakona0, 2022-06-24:
I also noticed that games will continue to run just fine if you uninstall the `amdvlk` package

tycho, 2022-06-24:
> > @raupach @tycho @mihalycsaba what version of Vulkan loader is installed in your system?
> 
> ```
> pacman -Qi vulkan-icd-loader
> Name            : vulkan-icd-loader
> Version         : 1.3.217-1
> ```

Same version of the loader here.

I'm pretty sure something changed in the implicit `VK_LAYER_AMD_switchable_graphics` layer which is causing the `vkEnumeratePhysicalDevices` behavior change.

Flakebi, 2022-06-24:
For what it’s worth, 2022.Q2.3 works fine on NixOS with vulkan-loader 1.3.211 and 1.3.216.

tycho, 2022-06-27:
I suspect it's only reproducible for people with switchable graphics (e.g. AMD iGPU + NVIDIA dGPU laptops).

gdiShun, 2022-06-28:
Had this issue with an AMD 6800, no iGPU. It also caused vulkaninfo to just hang.

baptisterajaut, 2022-06-28:
I second @gdiShun ! RX6900XT , vulkaninfo hangs ! vkcube did work tho. removed amdvlk and its 32 bits and it works.

Slater91, 2022-07-01:
> All see this issue on arch? Is anyone from non-arch distro that is affected?

I can reproduce on KDE neon (based on Ubuntu 20.04). I can't find a "vulkan-loader" or similar package, but vulkan-tools is at version 1.2.131.1. On version 2022.Q2.3 no games start, as they complain they can't find a DirectX 11-compatible device; reverting back to 2022.Q2.2 fixes the issue. I am on a desktop PC with a Radeon RX 5700 XT as the only GPU. I'm happy to provide more info if needed/requested.

mwip, 2022-07-16:
Had the same issues with The Sims 4 and Rogue Legacy 2

While the latter was fixed by rolling back amdvlk to `2022.Q2.2`, Sims needed a bit more care:

```
uname -a
Linux andlang 5.18.11-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 12 Jul 2022 15:44:35 +0000 x86_64 GNU/Linux

cd /var/cache/packman/pkg

sudo pacman -U vulkan-radeon-22.1.1-1-x86_64.pkg.tar.zst \
    lib32-vulkan-radeon-22.1.1-1-x86_64.pkg.tar.zst \
    vulkan-headers-1:1.3.213-1-any.pkg.tar.zst \
    lib32-vulkan-icd-loader-1.3.213-1-x86_64.pkg.tar.zst \
    vulkan-icd-loader-1.3.213-1-x86_64.pkg.tar.zst
```
Leaving this as it might help someone.

closed 2022-07-19 by jinjianrong