GPUOpen-Drivers/AMDVLK #283: Request: minimal VK_NV_device_generated_commands support for VKD3D (like RADV)..

oscarbg, 2022-07-18:
Hi,

I (we) want D3D12 ExecuteIndirect functionality for Vulkan..
at least is needed for VKD3D supporting games that make use of such functionality like Halo Infinite..

the closest VK support we have right now is from VK_NV_device_generated_commands ext.. which is in fact is as superset of D3D12 ExecuteIndirect it seems..

so VKD3D has merged support for ExecuteIndirect via that NV ext and Mesa's RADV ships support for a "minimally" VK_NV_device_generated_commands ext with enough features for VKD3D executeIndirect ([!17269 (merged)](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/17269))..

while we can wonder how far we are from a EXT or KHR version of VK_NV_device_generated_commands would be nice if in the meanwhile we get also "minimal support" from AMDVLK for VK_NV_device_generated_commands ext at least for making VKD3D happy..

I wish we get/expect similar VKD3D support from AMDVLK as from RADV..

thanks..

PS: I also opened similar request to Mesa Intel ANV developers so we get all Linux desktop VK driver support this functionality..

oscarbg, 2022-07-18:
by minimal I mean (see https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/17269#note_1451621)
`VK_INDIRECT_COMMANDS_TOKEN_TYPE_SHADER_GROUP_NV` isn't supported (see `maxGraphicsShaderGroupCount = 0`)
and also maxIndirectCommandsStreamCount is 1 vs NV 32

oscarbg, 2022-07-18:
on the positive side seems AMDVLK has been working on exposing ExecuteIndirect functionaly on Vulkan as pal commits like:
https://github.com/GPUOpen-Drivers/pal/commit/1ff58cafbae03aa571b0dde83aefd0d1fdfb8a9b
show lots of ExecuteIndirect packet support being worked on..

oscarbg, 2022-07-22:
Yep.. it's that bad for AMDVLK devs trying to work on it and expose it?

Degerz, 2024-06-05:
https://github.com/GPUOpen-Drivers/AMDVLK/issues/333

This issue can be closed as it is likely a duplicate related to above linked issue ...

jinjianrong, 2024-08-21:
The extension is exposed for vkd3d in 2024.Q2.2

closed 2024-08-21 by jinjianrong