GPUOpen-Drivers/AMDVLK #289: Request: support VK_NV_compute_shader_derivatives for VKD3D shader model 6.6 support

oscarbg, 2022-08-26:
Hi,

this extension is supported by Nvidia on recent GPUs and also on Mesa AMD RADV driver and Intel ANVIL driver..
it's required by VKD3D for exposing SM6.6 support:
https://github.com/HansKristian-Work/vkd3d-proton/blob/7160ee223a1966228ea925eb906517e5e542872d/libs/vkd3d/device.c#L6248
and it may be needed for running UE5 projects using nanite using VKD3D.. see:
https://github.com/HansKristian-Work/vkd3d-proton/issues/678#issuecomment-1080765645

thanks..

oscarbg, 2022-08-27:
note hope this is enabled also AMD Windows Vulkan driver..

oscarbg, 2022-10-21:
note latest Intel Windows driver 3729 includes it at least for Intel Xe and higher..

oscarbg, 2023-05-21:
Any update? Has work begun?

ruiminzhao, 2023-05-22:
This extension has already been delivered in commit-id: 97a755f58b7820c79d2f6315089fb70dd43df6a0
That means the binary after Feb can be used for this extension.
Thanks.

oscarbg, 2023-05-27:
Hi @ruiminzhao,
thanks for answering.. 
but seems still the extension not shipping in latest 2023q2.2 since I search for the extension and is missing everywhere:
1) this 6900xt vulkan.gpuinfo report using "2023.Q2.2 (LLPC)" driver doesn't show "VK_NV_compute_shader_derivatives" extension:
http://vulkan.gpuinfo.org/displayreport.php?id=21219#extensions
2) searching "VK_NV_compute_shader_derivatives" string is missing in amdvlk.so from amdvlk_2023.Q2.2_amd64 package..
3) searching in both:
https://github.com/GPUOpen-Drivers/xgl/blob/dev/icd/api/strings/extensions.txt
https://github.com/GPUOpen-Drivers/xgl/blob/dev/icd/api/include/vk_extensions.h
VK_NV_compute_shader_derivatives is missing..
so sorry for reasking but are you sure extension has been shipped in amdvlk repo?
thanks..

ruiminzhao, 2023-05-29:
@oscarbg After a further check and syncing, it can be confirmed this extension has been implemented. But it still need some further tests here. Whether to deliver this NV extension, it will be decided by our project manager. Thanks.

oscarbg, 2023-07-07:
thanks @ruiminzhao, hope if not delivering this NV ext, it's because a KHR or EXT extension is being developed like for mesh shader..

jinjianrong, 2024-06-06:
The extension is enabled for vkd3d in v-2023.Q4.3

closed 2024-06-06 by jinjianrong